ucb-bar/chipyard #2: Readme tutorial outdated

bjoernveit, 2017-02-20:
I'm currently working on a university project. But im struggling with adding my one device to the Chip. 
For the project I need to get a pwm working, which is a nice coincidence.
Anyway with the current state of the Rocket-Chip, the tutorial is not working.
I understand that the rocket-chip is transitioning from TileLinkIO to TileLinkIO2 atm, so it's probably not the best idea to updated the Tutorial right now.
But like I said we are struggling to make it work, so any help or information would be very much appreciated.

mwachs5, 2017-02-20:
I'm not sure about this tutorial, but have you looked at other example projects using the current rocket-chip? You could see

https://github.com/sifive/sifive-blocks/blob/master/src/main/scala/devices/pwm/PWMPeriphery.scala

That example uses rocket-chip's TLRegisterRouter to make a simple register-controlled device (TLPWM), then uses Periphery traits to hook it up into the overall SoC. This is basically what this tutorial describes, but with updated syntax that is known to work with current rocket-chip.

emantoni, 2017-03-20:
I have problem to understand the process of how to add an accelerator!Can anyone help me and tell me the steps i have to do, the files i have to change?

bjoernveit, 2017-03-20:
In the current state of the rocketchip, it does not work out of the box. You can read about it in the issue [here.](https://github.com/seldridge/rocket-rocc-examples/issues/2)
The repo itself tells you how to work around the problems though. It will show you how to get the build in accelerators to work. https://github.com/seldridge/rocket-rocc-examples/
Note that the repository tells you to use an older commit of the rocket-chip repository, but you can actually use the version pointed to by this project-template repository. 
Once you get the build in once to run you can try implementing your own instead, the RoCC interface is actually documented pretty well [here](https://docs.google.com/document/d/1CH2ep4YcL_ojsa3BVHEW-uwcKh1FlFTjH_kg5v8bxVw/edit). 

You'll find the build in accelerators here `rocket-chip/src/main/scala/rocket/rocc.scala`, other then that you only have to change the mapping of the custom0-3 opcodes to your Accelerator here: `rocket-chip/src/main/scala/coreplex/Configs.scala`

I myself have some problems getting my own one to run, so so far i edited the build in ones to fit my purpose, so you might one to fall back to that, too. Other then that i wish you the best of luck, you'll need it ;).

zhemao, 2017-07-08:
I've updated this repo to track (close to) the latest master of rocketchip and added better documentation. Hopefully things should be more understandable now.

closed 2017-07-08 by zhemao