ucb-bar/chipyard #3: Segfault when executing example PWM programme

LucasKl, 2017-03-10:
Hello,

i followed the tutorial and succesfully compiled the verilator emulator with the PWM.
However running the emulator with the example code always leads to a segfault.

../emulator/emulator-rocketchip-PWMTLConfig ../riscv-tools/riscv-pk/build/pk pwm
z  0000000000000000 ra 00000000000101b8 sp 000000000feeeb10 gp 0000000000011d30
tp 0000000000000000 t0 0000000000011598 t1 0000000000011598 t2 0000000000000000
s0 000000000feeeb40 s1 0000000000000000 a0 0000000000002000 a1 0000000000000014
a2 0000000000000000 a3 0000000000000000 a4 0000000000000014 a5 0000000000002000
a6 0000000000000000 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 0000000000010188 va 0000000000002000 insn       ffffffff sr 8000000000006000
User store segfault @ 0x0000000000002000
*** FAILED *** (code = -1, seed 1489144863) after 2329415 cycles

Is there anybody who had the same issue already and might point me in the right direction? 
Any help is very appreciated.

Thanks,
Lucas

zhemao, 2017-03-10:
The pwm program doesn't use pk. It's meant to run baremetal.

LucasKl, 2017-03-12:
Ok thank you for helping me here. How can i write to the pwm registers with the proxy kernel? I tried writing to /dev/men but that doesn't seem to be working.

Thanks,
Lucas

ViccoYang, 2017-04-06:
Hello,i followed the tutorial but have problem compiling the verilator emulator. When I run make in the "verisim" directory, it shows sbt.resolveexception unresolved dependency edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT
Have you met this problem? Can you help me with that? Thanks.

bjoernveit, 2017-04-11:
You need to go in the firrtl folder and follow the README there.
It's important that you not only build it but also run something like `sbt publish local` (not sure if this is the exact command, pls check the README)

closed 2017-04-26 by zhemao