ucb-bar/chipyard #7: How to know how many cycles the program takes?

ViccoYang, 2017-04-12:
I follow the tutorial and run pwm.riscv successfully, now I wonder how to know how many cycles the program takes?
To go further, can I know how many cycles each function takes?
for example,If I write a sort.c to run on ./simulator-myproject-myConfig, and sort.c contains DataPrepare(),Sort() and DataCompare() 3 functions in main(), can I get the cycles cost for each function?

Thanks for your help.

ViccoYang, 2017-04-12:
[image]
I learned SOC Designer last year and as the pic shows, selectSort takes 6259cycles. Is there a similar way in RISCV tools?

zhemao, 2017-04-26:
You can use the RISC-V "rdcycle" instruction to get the current number of cycles.

    unsigned long read_cycles(void)
    {
        unsigned long cycles;
        asm volatile ("rdcycle %0" : "=r" (cycles));
    }

    int main(void)
    {
        unsigned long start, end;
        start = read_cycles();
        run_program();
        end = read_cycles();
        printf("Took %lu cycles\n", end - start);
    }

ViccoYang, 2017-04-26:
Thanks for your help but it seems doesn't work.
```
#define Reg1 0x2000
#include <stdio.h>
unsigned long read_cycles(void)
{
    unsigned long cycles;
    asm volatile ("rdcycle %0" : "=r" (cycles));
}
static inline void write_reg(unsigned long addr, unsigned long data)
{
	volatile unsigned long *ptr = (volatile unsigned long *) addr;
	*ptr = data;
}
static inline unsigned long read_reg(unsigned long addr)
{
	volatile unsigned long *ptr = (volatile unsigned long *) addr;
	return *ptr;
}
int main(void)
{
	unsigned long start, end;
	start = read_cycles();
	int j = 0,i = 0;
	write_reg(Reg1, 10000);
	int loopNum=read_reg(Reg1);
	/*for(i=0;i<loopNum;i++){
		j=j+i;
	}*/
	printf("%d %d\n",i,j);
	end = read_cycles();
	printf("Took %lu cycles and end %lu start %lu\n",end - start,end,start);
	return 0;
}
```
I run these code and get the result:
```
i is 0 and j is 0
Took 0 cycles and end 0 start 0
```
but if I run with +verbose it will show the total cycles
`Completed after 16686 cycles`

If I uncomment the loop inside and run again, the result is
```
i is 10000 and j is 49995000
Took 0 cycles and end 0 start 0
```
and
`Completed after 47509 cycles`

If I reduce 10000 to 5000 then the result is 
```
i is 5000 and j is 12497500
Took 0 cycles and end 0 start 0
```
and
`Completed after 32455 cycles`

Then maybe I can guess that 10000 loop takes 47509-16686=30823 cycles
then each loop takes about 3 cycles. When I only loop 5000 times
the total cycle maybe 16686+5000*3.0832=32102, which is very close to 32455,
then my guess looks reasonable.

The problem of this method is that every time when I run, the random seed is different
and I'm not sure how to proved that the seed will not affect the result too much.

So, the best and convincing way should be using "rdcycle" instruction inside the code but it always shows 0. I'm confused about this.
Can you give me some advise? Thanks for your help.

zhemao, 2017-04-26:
Oh sorry, I made a mistake in the code. You have to actually return "cycles".

    unsigned long read_cycles(void)
    {
        unsigned long cycles;
        asm volatile ("rdcycle %0" : "=r" (cycles));
        return cycles;
    }

zhemao, 2017-07-08:
Does my last example work? If so, I'd like to close this issue.

cokeco, 2018-01-30:
I have tried last example. It can actually work.

closed 2018-02-23 by zhemao

nenjeff, 2018-04-16:
Hello,

I have some question about this issue.
(1). I use the method above (RISC-V "rdcycle" instruction) to measure how many cycles the program takes.
And it gets:
[image]

(2). I have also checked out with .vcd file and make sure above method get correct result.
It complete in 86975(ns), and the clock period is 2(ns).
Therefor, the program takes about 43487 cycles to complete.
[image]

Why the results differ dramatically?  Or there is something I misunderstand?