ucb-bar/chipyard #14: Not a bug, but why use import "freechips.rocketchip.xxx" when rocket chip uses just xxx?

pkimelman-nxp, 2017-07-17:
This makes it hard to diff your version of common files from rocket, coreplex, config, etc. I am unclear why this extra name specificity when I would assume you could just default this? 
Also, why do you use "chip" vs. "rocketchip", and the better named "amba" vs. "uncore". I can understand that last one, but is it worth having a different name for the same files?

colinschmidt, 2017-07-17:
rocket-chip recently refactored its package hierarchy so it also now uses these new names "freechips.rocketchip" and "amba"

pkimelman-nxp, 2017-07-18:
Ah, OK. Thanks. I guess I have a split in time between them.

Regards, Paul

Paul Kimelman – Automotive Platform Architecture
408.906.9676

closed 2017-07-18 by colinschmidt