ucb-bar/chipyard #15: PWMBase module is instanced in .fir file but not exisiting in final rtl codes ? why

etchen66, 2017-08-08:
module PWMTL_pwm : 
    input clock : Clock
    input reset : UInt<1>
    output io : {interrupts : {}, flip in : {0 : {a : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<7>, address : UInt<14>, mask : UInt<4>, data : UInt<32>}}, flip b : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<7>, address : UInt<14>, mask : UInt<4>, data : UInt<32>}}, c : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<3>, size : UInt<2>, source : UInt<7>, address : UInt<14>, data : UInt<32>, error : UInt<1>}}, flip d : {flip ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<2>, size : UInt<2>, source : UInt<7>, sink : UInt<1>, addr_lo : UInt<2>, data : UInt<32>, error : UInt<1>}}, e : {flip ready : UInt<1>, valid : UInt<1>, bits : {sink : UInt<1>}}}}, pwmout : UInt<1>}
    
    clock is invalid
    reset is invalid
    io is invalid
    reg period : UInt<32>, clock @[PWM.scala 49:19]
    reg duty : UInt<32>, clock @[PWM.scala 51:17]
    reg enable : UInt<1>, clock with : (reset => (reset, UInt<1>("h00"))) @[PWM.scala 53:23]
    inst base of PWMBase @[PWM.scala 55:20]  
 
// comments :  in  lines above , PWMbase shall be instanced.  however, at the end , PWMBase module is totally removed, I don't know why , I just follow the instruction steps make PROJECT=example Confing=  PWMConfig, nothing else. The result is PWMBase disappear 

Any one who have same situations that could share ?
    
    io.pwmout <= base.io.pwmout @[PWM.scala 56:13]

etchen66, 2017-08-12:
I debugged for a whole day. 
I noticed if pwmout is not pulled to TestHarness, then firrtl will optimize the rtl codes , which leads PWMBase is totally removed. 
I manually update the fir file and connect the pwmout to TestHarness to make every module present. 

Now, I have a question, how to pull pwmout in chisel/scala language ? 

class TestHarness(implicit val p: Parameters) extends Module {
  val io = IO(new Bundle {
    val success = Output(Bool())
    val pwmout = Output(Bool()) 
  })

  val dut = p(BuildTop)(p)
   dut.connectSimAXIMem()
   io.success := dut.connectSimSerial()
   io.pwmout  := dut.pwmout                     <======  it report error during make CONFIG=PWMConfig
}

Please help ........

Thanks,
Steven

colinschmidt, 2017-08-17:
@jackkoenig would know more about why firrtl is optimizing away the PWM.

For the error you are encountering now it would be helpful to see the output, what you've done seems reasonable if pwmout is a Bool

zhemao, 2018-02-24:
The problem was that the DontTouch annotations weren't being passed  to the FIRRTL compiler. I've fixed those in 073c16961e91dad1c119286dfc0e737730ae6f9c.

closed 2018-02-24 by zhemao