ucb-bar/chipyard #17: simulated block device build error

riscveval, 2017-08-14:
Hi, When I am trying to add any block for example run make PROJECT=blkdev CONFIG=BlockDeviceConfig in the "verisim" directory, something goes wrong. The same issue i am facing for create own project & MIMO.

[info] Running blkdev.Generator /nfs/iind/disks/i2r.disk.0001/RISCV_Evaluation/Free-chip_project/rocket-radha/project-template/verisim/generated-src blkdev TestHarness blkdev BlockDeviceConfig                                                                                                                  
[error] (run-main-0) java.lang.ClassNotFoundException: blkdev.Generator

colinschmidt, 2017-08-16:
I believe you only want to set the CFG_PROJECT

ninode, 2017-08-19:
Hell, I have the same issue. 

```
mkdir -p /home/alpha/project-template/verisim/generated-src
cd /home/alpha/project-template && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/alpha/project-template/rocket-chip/sbt-launch.jar "run-main blkdev.Generator  /home/alpha/project-template/verisim/generated-src blkdev TestHarness blkdev BlockDeviceConfig"
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Set current project to testchip-example (in build file:/home/alpha/project-template/)
[info] Running blkdev.Generator /home/alpha/project-template/verisim/generated-src blkdev TestHarness blkdev BlockDeviceConfig
[error] (run-main-0) java.lang.ClassNotFoundException: blkdev.Generator
java.lang.ClassNotFoundException: blkdev.Generator
	at java.lang.ClassLoader.findClass(ClassLoader.java:530)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
[error] (compile:runMain) Nonzero exit code: 1
[error] Total time: 1 s, completed Aug 19, 2017 4:51:21 AM
make: *** [/home/alpha/project-template/verisim/generated-src/blkdev.TestHarness.BlockDeviceConfig.fir] Error 1
```
Could this be related to sbt ? 

Thank you

colinschmidt, 2017-08-19:
I think some of the configuration names got changed. I believe you both probably want to build something like this:
```
make CONFIG=SimBlockDeviceConfig PROJECT=example
```

Does that work?

ninode, 2017-08-19:
Yes, that solves it. Thank you Colin.

colinschmidt, 2017-08-22:
No that should still be needed. Are you sure that file exists and is readable?

ninode, 2017-08-22:
@colinschmidt, here are the files which the verisim directory contains:

```
alpha@alpha-VirtualBox:~/project-template/verisim$ ls
generated-src  Makefrag-verilator                      verilator
Makefile       simulator-example-SimBlockDeviceConfig
```
There is no block-device.img file.

Also, the PWM has the same issue when running make PROJECT=pwm CONFIG=PWMConfig ?
 Did those names change as well ?

ninode, 2017-08-22:
@colinschmidt, here are the files which the verisim directory contains:

```
alpha@alpha-VirtualBox:~/project-template/verisim$ ls
generated-src  Makefrag-verilator                      verilator
Makefile       simulator-example-SimBlockDeviceConfig
```
There is no block-device.img file.

Also, the PWM has the same issue when running make PROJECT=pwm CONFIG=PWMConfig ?
```
[error] (run-main-0) java.lang.ClassNotFoundException: pwm.Generator
java.lang.ClassNotFoundException: pwm.Generator
```

 Did those names change as well ?

zhemao, 2017-08-22:
Hi sorry. I changed the names but forgot to update the README. It's fixed now.

The block device file is something you have to create yourself. If you want to run the blkdev.riscv program in tests, you can create a device as follows.

    dd if=/dev/zero of=block-device.img bs=512 count=4
    ./simulator-example-SimBlockDeviceConfig +blkdev=block-device.img ../tests/blkdev.riscv

ninode, 2017-08-22:
Thank you @zhemao. What about the pwm ? I got an error when I ran make PROJECT=pwm CONFIG=PWMConfig .

zhemao, 2017-08-22:
That's also been fixed. Check the new README.

closed 2017-08-22 by zhemao