ucb-bar/chipyard #18: Compiling with llvm and pk

ninode, 2017-08-23:
Hello, 

I'm compiling the pwm.c program using riscv64-unknown-elf-gcc -o pwm pwm.c and then pk pwm. But it's not working. Should I use LLVM instead ? With clang -target riscv64 -mriscv=RV64IAMFD -S pwm.c -o pwm.S and riscv64-unknown-elf-gcc -o pwm.riscv pwm.S ? 

Is it possible to use pk at all ? 

Thank you

colinschmidt, 2017-08-23:
That output doesn't look like you are using riscv-gcc. Are you sure its on your PATH?

ninode, 2017-08-24:
Again, thank you Colin. It solved it

closed 2017-08-24 by ninode

ninode, 2017-08-24:
Can I run pwm.c without make ? 

I did the following and got the pwm.riscv binary but I didn't use the linker. Is it wrong ?

```
riscv64-unknown-elf-gcc -mcmodel=medany -std=gnu99 -O2 -fno-common -fno-builtin-printf -Wall -c pwm.c -o pwm.o

riscv64-unknown-elf-objdump -D pwm.o > pwm.riscv
```