ucb-bar/chipyard #20: Problem when running PWMConfig

Ddnirvana, 2017-09-29:
Hi, guys,

I am new about chisel and rocket project so I am trying learn things about them by following this project-template. 
I can successfully run this project using DefaultExampleConfig and SimBlockDeviceConfig, but when I tried the PWMConfig, I failed in the `make CONFIG=PWMConfig` command.

The error is as the following: 

Could you please tell me how to fix this problem and what's cause?
Any advice will be helpful! Thanks a lot!

<pre>
info] Running example.Generator /home/dd/devlop/riscv/chisel/project-template/project-template/verisim/generated-src example TestHarness example PWMConfig
[info] [0.012] Elaborating design...
Interrupt map (2 harts 0 interrupts):

[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at testchipip.GeneratorApp$$anonfun$circuit$1.apply(Generator.scala:22)
        at testchipip.GeneratorApp$$anonfun$circuit$1.apply(Generator.scala:19)
        at chisel3.core.Module$.do_apply(Module.scala:42)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:240)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:238)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at chisel3.internal.Builder$.build(Builder.scala:238)
        at chisel3.Driver$.elaborate(Driver.scala:92)
        at testchipip.GeneratorApp$class.circuit(Generator.scala:19)
        at example.Generator$.circuit$lzycompute(TestHarness.scala:20)
        at example.Generator$.circuit(TestHarness.scala:20)
        at testchipip.GeneratorApp$class.generateFirrtl(Generator.scala:30)
        at example.Generator$.generateFirrtl(TestHarness.scala:20)
        at example.Generator$.delayedEndpoint$example$Generator$1(TestHarness.scala:21)
        at example.Generator$delayedInit$body.apply(TestHarness.scala:20)
        at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
        at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
        at scala.App$$anonfun$main$1.apply(App.scala:76)
        at scala.App$$anonfun$main$1.apply(App.scala:76)
        at scala.collection.immutable.List.foreach(List.scala:381)
        at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
        at scala.App$class.main(App.scala:76)
        at example.Generator$.main(TestHarness.scala:20)
        at example.Generator.main(TestHarness.scala)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
Caused by: java.lang.IllegalArgumentException: requirement failed
at scala.Predef$.require(Predef.scala:207)
        at example.PWMTLModule$class.$init$(PWM.scala:46)
        at example.PWMTL$$anonfun$5$$anon$1.<init>(PWM.scala:75)
        at example.PWMTL$$anonfun$5.apply(PWM.scala:75)
        at example.PWMTL$$anonfun$5.apply(PWM.scala:75)
        at freechips.rocketchip.tilelink.TLRegisterRouter.module$lzycompute(RegisterRouter.scala:145)
        at freechips.rocketchip.tilelink.TLRegisterRouter.module(RegisterRouter.scala:145)
        at freechips.rocketchip.tilelink.TLRegisterRouter.module(RegisterRouter.scala:128)
        at freechips.rocketchip.diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:61)
        at freechips.rocketchip.diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:61)
        at chisel3.core.Module$.do_apply(Module.scala:42)
        at freechips.rocketchip.diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:61)
        at freechips.rocketchip.diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:58)
        at scala.collection.immutable.List.foreach(List.scala:381)
        at freechips.rocketchip.diplomacy.LazyModule.instantiate(LazyModule.scala:58)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike$class.$init$(LazyModule.scala:154)
        at freechips.rocketchip.diplomacy.LazyMultiIOModuleImp.<init>(LazyModule.scala:161)
        at freechips.rocketchip.coreplex.BareCoreplexModule.<init>(BaseCoreplex.scala:25)
        at freechips.rocketchip.coreplex.BaseCoreplexModule.<init>(BaseCoreplex.scala:91)
        at freechips.rocketchip.coreplex.RocketCoreplexModule.<init>(RocketCoreplex.scala:129)
        at example.ExampleTopModule.<init>(Top.scala:18)
        at example.ExampleTopWithPWMModule.<init>(Top.scala:31)
        at example.ExampleTopWithPWM.module$lzycompute(Top.scala:27)
        at example.ExampleTopWithPWM.module(Top.scala:27)
        at example.WithPWM$$anonfun$$lessinit$greater$2$$anonfun$apply$2$$anonfun$applyOrElse$2$$anonfun$apply$6.apply(Configs.scala:16)
        at example.WithPWM$$anonfun$$lessinit$greater$2$$anonfun$apply$2$$anonfun$applyOrElse$2$$anonfun$apply$6.apply(Configs.scala:16)
        at chisel3.core.Module$.do_apply(Module.scala:42)
        at example.WithPWM$$anonfun$$lessinit$greater$2$$anonfun$apply$2$$anonfun$applyOrElse$2.apply(Configs.scala:16)
        at example.WithPWM$$anonfun$$lessinit$greater$2$$anonfun$apply$2$$anonfun$applyOrElse$2.apply(Configs.scala:15)
        at example.TestHarness.<init>(TestHarness.scala:15)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at testchipip.GeneratorApp$$anonfun$circuit$1.apply(Generator.scala:22)
        at testchipip.GeneratorApp$$anonfun$circuit$1.apply(Generator.scala:19)
        at chisel3.core.Module$.do_apply(Module.scala:42)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:240)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:238)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at chisel3.internal.Builder$.build(Builder.scala:238)
        at chisel3.Driver$.elaborate(Driver.scala:92)
        at testchipip.GeneratorApp$class.circuit(Generator.scala:19)
        at example.Generator$.circuit$lzycompute(TestHarness.scala:20)
        at example.Generator$.circuit(TestHarness.scala:20)
        at testchipip.GeneratorApp$class.generateFirrtl(Generator.scala:30)
        at example.Generator$.generateFirrtl(TestHarness.scala:20)
        at example.Generator$.delayedEndpoint$example$Generator$1(TestHarness.scala:21)
        at example.Generator$delayedInit$body.apply(TestHarness.scala:20)
        at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
        at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
        at scala.App$$anonfun$main$1.apply(App.scala:76)
        at scala.App$$anonfun$main$1.apply(App.scala:76)
        at scala.collection.immutable.List.foreach(List.scala:381)
        at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
        at scala.App$class.main(App.scala:76)
        at example.Generator$.main(TestHarness.scala:20)
        at example.Generator.main(TestHarness.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
        at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
[error] (compile:runMain) Nonzero exit code: 1
[error] Total time: 6 s, completed Sep 29, 2017 2:35:45 PM

</pre>

hauhsu, 2017-12-10:
I have the same problem.
It seems the requirement failed in pwm.scala:

``` scala
trait PWMTLModule extends Module with HasRegMap {
  val io: PWMTLBundle
  implicit val p: Parameters
  def params: PWMParams

  val w = params.beatBytes * 8
  require(w <= 32)  //FAILED HERE
  ...
}
```

I am new to Scala/Chisel and I can't figure out how does this code work.
It seem that `PWMTLModule` defines an abstract function `params` but not implement it?
And if I assign `w` a fixed number, say, 4, it will compile.
Can anyone give some advices about this problem? 
Thanks a lot!

colinschmidt, 2017-12-11:
Sounds like it might be a mismatch between beatBytes coming from rocket-chip and what the PWM module wants.

Are you just using master of this repo?

hauhsu, 2017-12-11:
Yeap, I just using the master branch and its submodules.

hauhsu, 2017-12-13:
I think the problem is that, when XLen is configured as 64-bit, beatBytes equals to 8.
But here w is expected to be smaller than 32, so the requirement failed.
If I modify the code as follows, it will work: 

``` scala
trait PWMTLModule extends HasRegMap {
  val io: PWMTLBundle
  implicit val p: Parameters
  def params: PWMParams

  val w = params.beatBytes * 8
  require(w <= 64)

  // ...
  
  //modify register map to avoid address overlap
  regmap(
    0x00 -> Seq(
      RegField(w, period)),
    0x08 -> Seq(
      RegField(w, duty)),
    0x016 -> Seq(
      RegField(1, enable)))
}
```

lu-ping, 2017-12-17:
Using 64bit register for PWM is waste and totally unneccessary.
my solution for this is hard code it in PWM.scala

$ git diff src/main/scala/example/PWM.scala
diff --git a/src/main/scala/example/PWM.scala b/src/main/scala/example/PWM.scala
index 3a8ebf7..64df19f 100644
--- a/src/main/scala/example/PWM.scala
+++ b/src/main/scala/example/PWM.scala
@@ -80,7 +80,7 @@ trait HasPeripheryPWM extends HasPeripheryBus {
   private val address = 0x2000

   val pwm = LazyModule(new PWMTL(
-    PWMParams(address, pbus.beatBytes))(p))
+    PWMParams(address, 32))(p))

   pwm.node := pbus.toVariableWidthSlaves
 }

An over look solution should be create a new config key for the periphery register width. but for me 32 is just enough.

zhemao, 2018-02-23:
Sorry I didn't address this issue until now. I'm just going to fix the width at 32 so that we don't have this issue of changing bus widths.

closed 2018-02-24 by zhemao