ucb-bar/chipyard #21: What is the purpose of the bootrom?

ccelio, 2017-11-05:
TL,DR: Is project-template/bootrom a testchipip construct? 

I'm wondering what is the difference between the bootrom in top-level project-template and rocket-chip. 

AFAICT, Rocket-chip's bootrom is built to use its rocket-chip/csrc/SimDTM.cc interface. It seems the project-template bootrom is build for the SimSerial from testchipip? If that is the case, that seems to suggest the project-template/bootrom should be moved to the testchipip repository. 

Or is it more generic?

zhemao, 2018-02-23:
Followed up with Chris in person on this, but commenting for the record.

ProjectTemplate uses the Tethered Serial Interface from TestChipIP to load the program into the simulation, unlike RocketChip, which uses the debug port. Therefore, the bootrom program for ProjectTemplate needs to be different.

closed 2018-02-23 by zhemao

ccelio, 2018-02-23:
My point is project-template shouldn't be the place to store the bootrom; I think the bootrom should be moved into testchipip since that's what it is used with.