ucb-bar/chipyard #23: Failed to compile c program in tests directory with 32bit configuration

nnamenoname, 2018-01-13:
Hi,
I set XLEN=32 for whole system, so I intend to run 32 bit ELF on that system.
I just modify Makefile(in project-template/tests/)configuration to generate 32 bit ELF file.

revision part:
GCC=riscv32-unknown-elf-gcc
OBJDUMP=riscv32-unknown-elf-objdump

I only revise two statements shown above, and remain part still as original version.
However, I get errors when I utilize new Makefile.
error message:
riscv32-unknown-elf-gcc -T link.ld -static -nostdlib -nostartfiles -lgcc pwm.o crt.o syscalls.o -o pwm.riscv
syscalls.o: In function `.L48':
syscalls.c:(.text+0x13c): undefined reference to `__umoddi3'
syscalls.o: In function `.L59':
syscalls.c:(.text+0x168): undefined reference to `__udivdi3'
syscalls.c:(.text+0x180): undefined reference to `__umoddi3'
collect2: error: ld returned 1 exit status
make: *** [pwm.riscv] Error 1

It fails in linking stage. Obviously, it seems there are some program in systemcall.c with 32bit configuration.
Can anyone tell me how to fix it for compiling successfully?

codelec, 2018-01-13:
```bash
./configure --prefix=$RISCV --enable-multilib
make -j4
```
You will have riscv64-unknown-elf-gcc installed to which if you pass `-march=rv32i -mabi=ilp32` will generate a binary complying to rv32i or for other options you can refer to https://www.sifive.com/blog/2017/08/14/all-aboard-part-1-compiler-args/  blog post

nnamenoname, 2018-01-14:
I fix the issue, thanks.

closed 2018-01-14 by nnamenoname

aignacio, 2018-11-20:
Hello @codelec,

Seems that this issue persists even with *--enable-multilib*.....
```bash
/home/aignacio/riscv_multilib/bin/riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -mcmodel=medany -std=gnu99 -O2 -fno-common -fno-builtin-printf -Wall -save-temps=obj -Iinc -c src/rocc_ha.c -o obj/rocc_ha.o
Linking...output/mm.baremetal
/home/aignacio/riscv_multilib/bin/riscv64-unknown-elf-gcc -Tld/link.ld -static -nostdlib -nostartfiles -lgcc -march=rv32imac -mabi=ilp32 -mcmodel=medany -std=gnu99 -O2 -fno-common -fno-builtin-printf -Wall -Iinc obj/crt.o obj/syscalls.o obj/mm.o obj/rocc_ha.o -o output/mm.baremetal
/home/aignacio/riscv_multilib/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: obj/syscalls.o: in function `.L48':
syscalls.c:(.text+0x102): undefined reference to `__umoddi3'
/home/aignacio/riscv_multilib/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: obj/syscalls.o: in function `.L59':
syscalls.c:(.text+0x122): undefined reference to `__udivdi3'
/home/aignacio/riscv_multilib/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: syscalls.c:(.text+0x132): undefined reference to `__umoddi3'
collect2: error: ld returned 1 exit status
Makefile:32: recipe for target 'output/mm.baremetal' failed
make: *** [output/mm.baremetal] Error 1

```