ucb-bar/chipyard #25: Failed to generate emulator with the newest master branch version

RaulGoal, 2018-01-17:
Hi, I clone the newest version from master branch at 2017.1.17.
Then I follow the README.md part to build emulator. However, it fails with some errors happen.
error message:
/project-template/testchipip/csrc/SimSerial.cc: In function ‘int serial_tick(unsigned char, unsigned char*, int, unsigned char*, unsigned char, int*)’:
/project-template/testchipip/csrc/SimSerial.cc:26:87: error: no matching function for call to ‘tsi_t::tsi_t(std::vector<std::basic_string<char> >)’
         tsi = new tsi_t(std::vector<std::string>(info.argv + 1, info.argv + info.argc));
                                                                                       ^
/project-template/testchipip/csrc/SimSerial.cc:26:87: note: candidate is:
In file included from /project-template/testchipip/csrc/SimSerial.cc:5:0:
/project-template/rocket-chip/riscv-tools/include/fesvr/tsi.h:21:3: note: tsi_t::tsi_t(int, char**)
   tsi_t(int argc, char** argv);
   ^
/project-template/rocket-chip/riscv-tools/include/fesvr/tsi.h:21:3: note:   candidate expects 2 arguments, 1 provided
make[1]: *** [SimSerial.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/project-template/testchipip/csrc/verilator-harness.cc: In function ‘int main(int, char**)’:
/project-template/testchipip/csrc/verilator-harness.cc:84:66: error: no matching function for call to ‘tsi_t::tsi_t(std::vector<std::basic_string<char> >)’
   tsi = new tsi_t(std::vector<std::string>(argv + 1, argv + argc));
                                                                  ^
/project-template/testchipip/csrc/verilator-harness.cc:84:66: note: candidate is:
In file included from /project-template/testchipip/csrc/verilator-harness.cc:7:0:
/project-template/rocket-chip/riscv-tools/include/fesvr/tsi.h:21:3: note: tsi_t::tsi_t(int, char**)
   tsi_t(int argc, char** argv);
   ^
/project-template/rocket-chip/riscv-tools/include/fesvr/tsi.h:21:3: note:   candidate expects 2 arguments, 1 provided
make[1]: *** [verilator-harness.o] Error 1
make[1]: Leaving directory `/project-template/verisim/generated-src/example.TestHarness.PWMConfig'
make: *** [/project-template/verisim/simulator-example-PWMConfig] Error 2

Does anyone successfully build out the emulator with newest version from master branch?
Could anyone helps me about it?

cokeco, 2018-01-19:
I have encountered same problem when run make CONFIG=PWMConfig.
Does anyone fix the problem?

seldridge, 2018-01-19:
It looks like this is me: https://github.com/riscv/riscv-fesvr/pull/25

There was a change to the way that arguments are exposed backwards from HTIF that changed this API. It looks like project template is using an older version of testchipip that's incompatible.

I _think_ that all that has to happen is a bump to testchipip.

seldridge, 2018-01-19:
It looks like testchipip will also need modifications to its verilator-harness.cc. That could also benefit from an update to act like rocket-chip's emulator.cc. I may be able to look at this tomorrow.

RaulGoal, 2018-02-02:
verilator-harness.cc has been revised for HTIF/FESVR API changes

closed 2018-02-02 by RaulGoal