ucb-bar/chipyard #27: [Not a bug] Why it costs lots of time when rebuild out the emulator?

cokeco, 2018-03-06:
I rebuild the emulator with some revisions.  I rebuild the project again without typing "make clean" before.
It takes about first time.
I use make -d  to observe the hanging point that costs lots of time at rebuild.

hanging point:
```
The prerequisites of 'default' are being made.
Live child 0xaddc20 (VTestHarness.o) PID 31961
```
It seems it recompile to generate out VTestHarness.o(I am not sure) .
I cannot figure out why? Could anyone tells me the reasons?

olix86, 2018-03-25:
I would also be interested if someone knows the reason! I've noticed the same phenomenon, and it would be great to be able to make minor changes to a configuration without rebuilding the entire thing!

zhemao, 2018-05-10:
This is an unfortunate limitation of how verilator works. When you change the verilog in any way, verilator will create an entirely new set of C++ files, so they will all get rebuilt.

closed 2018-05-10 by zhemao