ucb-bar/chipyard #30: Question about measuring how many cycles the program takes with RISC-V "rdcycle" instruction.

nenjeff, 2018-04-16:
Hello,

I have referenced issue #7.
(1). I use the method RISC-V "rdcycle" instruction to measure how many cycles the program takes.
And it gets:
[image]

(2). I have also checked out with .vcd file and make sure above method get correct result.
It complete in 86975(ns), and the clock period is 2(ns).
Therefor, the program takes about 43487 cycles to complete.
image
[image]

Why the results differ dramatically? Or there is something I misunderstand?

zhemao, 2018-05-03:
Are you saying that the simulation terminated 43487 cycles in?

The point at which the rdcycle instruction executes is long before the simulation terminates.

nenjeff, 2018-05-16:
"Are you saying that the simulation terminated 43487 cycles in?"
->Yes

closed 2020-04-28 by alonamid