ucb-bar/chipyard #31: How to add a custom peripherical signal in the class TestHarness?

gvillanovanm, 2018-04-17:
I made the pwm example, but now I want to get the peripherical signal in the TestHarness, how to do it?
``` 
class TestHarness(implicit val p: Parameters) extends Module {
  val io = IO(new Bundle {
    val success = Output(Bool())
    val output_perip = Output(Bool())
  })
  val dut = p(BuildTop)(clock, reset.toBool, p)
  dut.debug := DontCare
  dut.connectSimAXIMem()
  dut.dontTouchPorts()
  dut.tieOffInterrupts()
  io.success := dut.connectSimSerial()

  io.output_perip := io_pwmout // ?? Here 
}
```
I wanted my io.output_perip in this module: 
[image]

Thanks!

zhemao, 2018-05-10:
You likely want to do something like this:

    io.output_perip := dut.asInstanceOf[ExampleTopWithPWMModule].io.pwmout

gvillanovanm, 2018-05-24:
Yes, it works fine now! Thanks.

closed 2018-05-24 by gvillanovanm