ucb-bar/chipyard #37: Adding JTAG module

aignacio, 2018-11-03:
Hello all,
I'm trying to add JTAG debug module to the default example to use remote bitbang (debug with GDB) but when I try to build the emulator some error about **jtag_tick** shows in the terminal with "make" or "make debug" like the following one:
```bash
g++    SimBlockDevice.o SimSerial.o blkdev.o verilator-harness.o verilated.o verilated_dpi.o verilated_vcd_c.o VTestHarness__ALL.a   -L/home/aignacio/riscv-tools/lib -Wl,-rpath,/home/aignacio/riscv-tools/lib -L/home/aignacio/build/demo/verisim -lfesvr -lpthread  -o /home/aignacio/build/demo/verisim/simulator-example-DefaultExampleConfig-debug -lm -lstdc++ 
VTestHarness__ALL.a(VTestHarness___024unit.o): In function `VTestHarness___024unit::__Vdpiimwrap_jtag_tick_TOP____024unit(unsigned char&, unsigned char&, unsigned char&, unsigned char&, unsigned char, unsigned int&)':
VTestHarness___024unit.cpp:(.text+0x89): undefined reference to `jtag_tick'
collect2: error: ld returned 1 exit status
VTestHarness.mk:76: recipe for target '/home/aignacio/build/demo/verisim/simulator-example-DefaultExampleConfig-debug' failed
make[1]: *** [/home/aignacio/build/demo/verisim/simulator-example-DefaultExampleConfig-debug] Error 1
make[1]: Leaving directory '/home/aignacio/build/demo/verisim/generated-src/example.TestHarness.DefaultExampleConfig.debug'
Makefile:66: recipe for target '/home/aignacio/build/demo/verisim/simulator-example-DefaultExampleConfig-debug' failed
make: *** [/home/aignacio/build/demo/verisim/simulator-example-DefaultExampleConfig-debug] Error 2
```
I modified the TestHarness class with the following code:
```bash
#I ommited the other parts because it's actually the same thing
import freechips.rocketchip.devices.debug.Debug
...
...
...
class TestHarness(implicit val p: Parameters) extends Module {
  val io = IO(new Bundle {
    val success = Output(Bool())
  })

  val dut = p(BuildTop)(clock, reset.toBool, p)
//   dut.debug := DontCare
  dut.reset := reset.toBool || dut.debug.ndreset
  dut.connectSimAXIMem()
  dut.dontTouchPorts()
  dut.tieOffInterrupts()
  io.success := dut.connectSimSerial()
  Debug.connectDebug(dut.debug, clock, reset.toBool, io.success)
}
```
and in the *packge example*:
```bash
# Configs.scala
import freechips.rocketchip.subsystem.{WithJtagDTM, WithRoccExample, WithNMemoryChannels, WithNBigCores, WithRV32}
...
...
...
class BaseExampleConfig extends Config(
  new WithBootROM ++
  new WithJtagDTM ++
  new freechips.rocketchip.system.DefaultConfig)
```
I'm using the same pattern as present in [TestHarness from RocketChip repo](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/system/TestHarness.scala) and AFAIK the only difference it's the chisel3 package that now uses different casting for reset. Could someone please advice me how exactly I should add jtag interface into the example? Thanks!

jimmysitu, 2018-11-24:
@aignacio,

jtag_tick is a DPI function in SimJTAG.cc, you may need to modify your makefile to ensure it is in the $sim_csrc list

https://github.com/freechipsproject/rocket-chip/blob/e9dc7d0fc95a84e6153e93b33223b5b22c6e7d5a/src/main/resources/csrc/SimJTAG.cc#L7

jimmysitu, 2018-11-24:
@aignacio 

jtag_tick is a DPI function in [SimJTAG.cc](https://github.com/freechipsproject/rocket-chip/blob/50bb13d7887e5f9ca192431234b057ae9d8edb6c/src/main/resources/csrc/SimJTAG.cc#L7-L14)
You may need to modify your makefile to ensure it is in the $sim_csrc list

closed 2019-09-28 by aignacio