ucb-bar/chipyard #39: Memory Address where TSI loads binary

muhammadali201, 2019-01-19:
At which address the Serial Device loads the program using TSI?
How does rocket-chip know the address to jump to after binary loading is finished?

abejgonzalez, 2019-06-02:
In the current `project-template` setup, the `bootrom` waits for the serial interface to load the program before executing it (see https://github.com/ucb-bar/testchipip/blob/master/src/main/resources/testchipip/bootrom/bootrom.S). It is specified in the `bootrom` that the start address of the program is at `0x80000000`.

abejgonzalez, 2019-10-13:
You can also reference this: https://chipyard.readthedocs.io/en/latest/Customization/Boot-Process.html. If your question is answered could you please close your issue!

closed 2019-10-14 by muhammadali201