ucb-bar/chipyard #89: Question about directly emit `HasPeripheryPWMAXI4ModuleImp` or `HasPeripheryPWMTLModuleImp`.

sequencer, 2019-05-18:
Instead of running a whole project of rocket-chip, I wonder if it is possible to just emit a minimal AXI4 Module for `PWM`, which is much easier to evaluate by Xilinx ZYNQ FPGAs or simply used as DUT for testing.

abejgonzalez, 2019-05-20:
So far as I know, since the mixin is a diplomatic mixin (i.e. uses `LazyModule` and connects to the `pbus`), it requires that you build it with the entire system. One thing that you could do is make a config to build the smallest possible Rocket core/uncore.

Maybe someone else knows a better way to do this?

sequencer, 2019-05-20:
Yes, I used to think using a "phantom" core to emit periphery only, however it's also too hard for me;(
I'm not familiar with the rocketchip ecosystem and has no idea where to start reading.

abejgonzalez, 2019-05-20:
The Berkeley folks here are trying to get some documentation up within the next couple of months to use this repo. However, Rocket-Chip is still a black hole of documentation. What I would try to do is just making your own small config or using a predone one like... https://github.com/freechipsproject/rocket-chip/blob/48b3e33c6cdb90bc9b86a7581c5be54f45c695e1/src/main/scala/system/Configs.scala#L59-L64

sequencer, 2019-05-20:
Thank you! I’m going to check this config. Trying to understand the essence of diplomacy.

abejgonzalez, 2019-05-28:
Seems like this is resolved for now. Will reopen if there are more questions or concerns.

closed 2019-05-28 by abejgonzalez