ucb-bar/chipyard #109: Docker images, installation instructions

jerryz123, 2019-06-01:
Several retreat attendees mentioned Docker images as a distribution format. I think this is a good idea.

At the very least, we should make the init scripts install all required packages that are not in a base Ubuntu install.

faddat, 2020-08-07:
I am working on this currently, my first attempt was with Arch Linux and I don't know if it's going to prove to be successful. If it is, I will send a pull request your way.

If it is not, I will do it with Ubuntu and also send a pull request your way.

I have been researching open CPU cores for about 5 months now and the first two maybe three months was spent getting the environment right, then learning Well, almost everything.

I started this as a very advanced Linux user without much knowledge of semiconductor design and engineering.  Actually, I still probably don't know that much about semiconductor design and engineering but I think that making systems like chipyard easier to use from the get-go will make these kinds of engineering tasks more approachable for more people and lead to a larger ecosystem of open source hardware.

faddat, 2020-08-08:
https://gitlab.com/virgohardware/core/-/blob/master/chipyard/Dockerfile

faddat, 2020-08-08:
It seems to be working steadily now though I do have some concerns about dependencies

faddat, 2020-08-10:
The current Dockerfile isn't appropriate for people who want to **run** shipyard.  

While I'll be using it in CI, I can't actually build an image from that Dockerfile, I get some version conflicts in the python packages.  

So I'm reworking it from bare Debian Sid.  

I've updated the Dockerfile referenced in #641

faddat, 2020-08-14:
In the discussion on #641, @jerryz123 and I talked about the proper form of a Chipyard Docker image.  

After several days of fiddling, I think my thoughts have changed.  Arch Linux has high quality precompiled binaries for the riscv toolchain, and they're kept up to date.  With https://gitlab.com/virgohardware/core I'm trying to make it as easy as possible to:

* modify / create an SOC
* build the SOC
* test the SOC
* compile Linux for the SOC
* run linux on it in simulation
* run linux on it using a cloud FPGA, probably using fire.sim
* download FPGA bitstreams for user-owned FPGAs
* prep the core for manufacturing and download the needed binaries for that

Why do I now favor Arch and precompiled RISCV Tools? 

It's the build times, and horsepower needed for a build.  

A complete image is going to contain:

* Chipyard
* Risc-v tools
* esp tools
* riscv-linux

....that's an incomplete list but those make up the majority of the build time.  I'd like for users to be able to use the exact same build environment as CI, so that there's no question about "if it builds on my PC or dev server, will it also build in the `core` CI system?"

@jerryz123 recommended to go with the versions released as submodules in Chipyard, as there could be compatibly issues.  

I am new-ish to RISC-V and semiconductors, but I am now thinking that it may be easier to get to this "end state goal" of fast dev env setups  weather building a Docker image locally or pulling a Docker image from our registry or somewhere else by testing:

currently:

My arch-based Dockerfile can build every core in `master`, no tests.
My ubuntu-based Dockerfile does not work, it dies during the riscv-tools setup

What I am going to pursue:

* Validate these functions on Arch:
  * Build
  * test in sim
  * build for fpga
  * build for manufacturing

Most open core projects and their HDLs have **great** documentation, but the development environment setup is both time consuming and complex.  

I currently have Arch-based Docker images for:

* Bluespec
* Chipyard
* SpinalHDL

Currently, I'm not using tools like `ccache` and maybe `distcc` though I was for a while and I plan to add them again.  

The feeling of creating an SOC from upstream modules and knowing that it could actually be built one day (though the 0.0.1 was not at all practical) was incredible!

I would like to help others do that (and more) without spending quite as much time on figuring out the development environment and `how to start`, and I have an inkling these Arch images will help because they'll drastically reduce setup times.  

but--

### what do you think?

:)

SaltedFishLZ, 2020-12-08:
BTW, with regard to the environment, according to the document on readthedocs.io, it seems on ubuntu we only need to install python3.6. But when I try to build the toolchain, I got an error that the command python is missing. I wonder the python used in the build scripts is python2 or python3?
I am using Ubuntu 18.04 server so it doesn't come with a Python 2.

closed 2021-03-29 by jerryz123