ucb-bar/chipyard #122: Test Suite Improvements

abejgonzalez, 2019-06-13:
Currently, `run-asm-tests` and `run-bmark-tests` run on `hartId == 0` of the system. This is supported for when either BOOM or Rocket is `hartId == 0`. However, we may want to think of a way to have asm/bmark tests that can target other `hartId`s (and maybe incorporate this in the `run` make functionality. This also applies to systems where there are different cores/accelerators ... BOOM + Rocket + Hwacha + Other.

jerryz123, 2019-06-13:
This seems pretty difficult to do, because the riscv-tests header files force the tests to only run on hartid 0.
It would be nice to have, though.

abejgonzalez, 2019-06-13:
While that is true, in this repo we do have control over the toolchain... so we could potentially add new tests, new headers, etc to make this possible. It is potentially tricky to do properly though.