ucb-bar/chipyard #146: run-binary-debug is confusing

jwright6323, 2019-07-12:
It's not clear that you need to set the BINARY variable. I'd recommend changing it to a pattern rule (e.g. `run-debug-%`). It's also not set as a `.PHONY` target, which is wrong. I'll PR this change next week.

jerryz123, 2019-07-12:
Will the pattern rule method work when the binary is not in the same directory?

jwright6323, 2019-07-12:
We can make something work

colinschmidt, 2019-07-12:
I think we want to take the approach of doing the simple things being easy and doing the hard things possible.
Running the tests bundled with chipyard that are in the right directory should be easy.
Running an executable located in some far away directory should be possible.

jerryz123, 2019-07-13:
I just do `make run-binary-debug BINARY=output/(longname)/(testname)`
Perhaps a `run-test-%` rule in addition to the `run-binary` rule would work?

jwright6323, 2020-06-09:
This is probably no longer needed

closed 2020-06-09 by jwright6323