ucb-bar/chipyard #147: Make -jN fails

jerryz123, 2019-07-13:
`make -j2` fails with 
```
[jerryz123@a5:/nscratch/jerryz/rebar2-proj/project-template/sims/vsim]$ make -j2
cd /nscratch/jerryz/rebar2-proj/project-template && java -Dsbt.ivy.home=/scratch/jerryz/.ivy2 -Dsbt.global.base=/scratch/jerryz/.sbt -Dsbt.boot.directory=/scratch/jerryz/.sbt/boot/ -Xmx16G -Xss16M -XX:MaxPermSize=256M -jar /nscratch/jerryz/rebar2-proj/project-template/generators/rocket-chip/sbt-launch.jar ++2.12.4 "project utilities" "runMain utilities.GenerateSimFiles -td /nscratch/jerryz/rebar2-proj/project-template/sims/vsim/generated-src/example.BoomRocketTestHarness.DefaultRocketConfig -sim vcs"
make: *** No rule to make target '/nscratch/jerryz/rebar2-proj/project-template/sims/vsim/generated-src/example.BoomRocketTestHarness.DefaultRocketConfig/example.BoomRocketTestHarness.DefaultRocketConfig.harness.mems.conf', needed by '/nscratch/jerryz/rebar2-proj/project-template/sims/vsim/generated-src/example.BoomRocketTestHarness.DefaultRocketConfig/example.BoomRocketTestHarness.DefaultRocketConfig.harness.mems.v'.  Stop.
make: *** Waiting for unfinished jobs....
```

jwright6323, 2019-07-13:
The Makefiles need to more explicitly list out all of the collateral that each step produces

jwright6323, 2019-07-19:
This will probably be fixed (or partially fixed) by @albert-magyar's barstools PR ucb-bar/barstools#63

PR #219 (2019-09-02): Make parallel support
This should now allow developers to run `make -jX` with the build system. It makes it so that a "temp" file is the sole target of the main recipe and the real targets depends on the "temp" target.

Followed: https://stackoverflow.com/questions/19822435/multiple-targets-from-one-recipe-and-parallel-execution

Fixes #147 and addresses part of #211.
closed 2019-10-13 by abejgonzalez