ucb-bar/chipyard #153: Stop calling everything "BoomAndRocketX"

jerryz123, 2019-07-15:


colinschmidt, 2019-07-15:
It does seem redundant if everything has that prefix

abejgonzalez, 2019-07-15:
"Blocket" system. https://github.com/riscv-boom/riscv-boom/pull/323

jwright6323, 2019-07-19:
I'm also not a fan of the long name, but serious question: what happens when we have a 3rd core and/or you start making `BoomSystolic` and `BoomHwacha` configs (without Rocket)? What about `RocketHbwif` and `BoomHbwif`, `RocketMyCoolWidget` and `BoomMyCoolWidget`, ... ?

abejgonzalez, 2019-07-19:
The long name was a short term solution to the naming problem. For the subsystem, I think it make a bit of sense. However for configs, I agree it gets too verbose. Maybe do what others do and just make it so that if you want boom and a systolic array you have the mixins and use them with the underscore to combine in the generator (a.k.a `make CONFIG=SmallBoomConfig_WithSystolicArray_WithWowWidget`)

jwright6323, 2019-07-19:
I like that, but if you have `CONFIG=SmallBoomAndRocketConfig_WithSystolicArray` what core gets the systolic array?

colinschmidt, 2019-07-19:
I think the underscores can be a solution but we need to make sure we can create reasonable options.

jwright6323, 2019-07-19:
I'm not super familiar with the `site, here, up` shenanigans, but can you make a config would attach to previously config'ed cores? e.g. something like (spitballing a bit):

`SmallBoomConfig_WithAnotherRocket`
=> Boom and Rocket
vs.
`SmallBoomConfig_WithSystolicArray_WithAnotherRocket`
=> (Boom + Systolic) and Rocket
vs.
`SmallBoomConfig_WithSystolicArray_WithAnotherRocket_WithHwacha`
=>  (Boom + Systolic) and (Rocket + Hwacha)
vs.
`SmallBoomConfig_WithSystolicArray_WithAnotherRocket_WithSystolicArray`
=>  (Boom + Systolic) and (Rocket + Systolic)

abejgonzalez, 2019-07-19:
Yes. You can do that.

jwright6323, 2019-07-19:
Which names?

abejgonzalez, 2019-07-19:
Sorry. I was not clear at all. Thinking about how people interact with the configs, we will probably have to change the output names for things such as the folders, simulator names since these will make the current system very clunky. We should also change the name to be something like how we currently compose things... `BaseConfig_WithLargeBoom_WithNormalHwacha_WithSmallRocket_WithLargeSystolic`.

abejgonzalez, 2019-08-30:
Solved if #211 is merged.

closed 2020-04-28 by alonamid