ucb-bar/chipyard #191: Add vsrc at location other than  "/generators/rocket-chip/src/main/resources/vsrc"

hitoswal, 2019-08-08:
I have created a subproject and have my own verilog files I want to use. But when I try to generate Verilog, it throws an error that 'files at the location mentioned in "addResource()" not found'. Can anybody please tell me how can I add vsrc directories in other subprojects as well?

Thanks!

abejgonzalez, 2019-08-08:
Hi! In addition to calling the `addResource` function with the file, you need to also change the `build.sbt` to add your project to the firrtl compiler classpath. Take a look at the following document... let me know if that helps: https://github.com/ucb-bar/chipyard/blob/abe-docs-dev/docs/Advanced-Usage/Resources.rst

hitoswal, 2019-08-08:
Thanks A lot. This helped :+1:

closed 2019-08-08 by hitoswal