ucb-bar/chipyard #198: Verilator giving error while compiling remote_bitbang.h

hitoswal, 2019-08-13:
I get the following error when Verilator starts compiling remote_bitbang.h
```
%Error: <path_to_repo>/sims/verisim/generated-src/<Build_Dir>/remote_bitbang.h:3: syntax error, unexpected '#'
%Error: <path_to_repo>/sims/verisim/generated-src/<Build_Dir>/remote_bitbang.h:6: Unsupported: Verilog 2001-config reserved word not implemented; probably you want `include instead: include
%Error: <path_to_repo>/sims/verisim/generated-src/<Build_Dir>/remote_bitbang.h:7: Unsupported: Verilog 2001-config reserved word not implemented; probably you want `include instead: include
%Error: <path_to_repo>/sims/verisim/generated-src/<Build_Dir>/remote_bitbang.h:9: Unsupported: SystemVerilog 2005 reserved word not implemented: class
%Error: <path_to_repo>/sims/verisim/generated-src/<Build_Dir>/remote_bitbang.h:31: syntax error, unexpected unsigned
%Error: <path_to_repo>/sims/verisim/generated-src/<Build_Dir>/remote_bitbang.h:41: Unsupported: Static in this context
%Error: <path_to_repo>/sims/verisim/generated-src/<Build_Dir>/remote_bitbang.h:41: syntax error, unexpected const, expecting TYPE-IDENTIFIER
%Error: Exiting due to 7 error(s)
```
Not sure whats happening.
Thanks!

colinschmidt, 2019-08-14:
What version of verilator do you have installed?

hitoswal, 2019-08-14:
`Verilator 4.008 2018-12-01 rev UNKNOWN_REV`
This what the script installed by default.

abejgonzalez, 2019-08-14:
Are you using the most recent version of Chipyard? It should be using `4.016`?

hitoswal, 2019-08-16:
I tried with Verilator version 4.016 as well. It fails. 
_FYI: I am using old version of riscv-tools._
I Tried building the latest version provided with this repo. It fails without giving error. I have repoted the related issue in _issue#203_

hitoswal, 2019-08-16:
One more thing I've observed is when there is JTAG in the Config I am generating, it gives above error but there there is DMI, it compiles successfully.

abejgonzalez, 2019-08-18:
So if I remember correctly... the `remote_bitbang.h` file is being included in the `$(build_dir)/sim_files.common.f`. Can you confirm if this is the case? You can just delete it from that for now.

hitoswal, 2019-08-21:
Yes, there is `remote_bitbang.h` included in `$(build_dir)/sim_files.common.f`. and once I remove the file, it compiles sucessfully.

abejgonzalez, 2019-08-21:
Sounds good. For now, go ahead and add something like `grep -v ".*\.h"` to filter out the `.h` files in the Makefile. We will push a patch to the `dev` branch with the fix. When Chipyard is next released this will be addressed. Thanks for bringing it up!

abejgonzalez, 2019-08-30:
Solved by #205 on the `dev` branch. Will be updated in the next `dev` -> `master` merge.

abejgonzalez, 2019-10-13:
This should be fixed now (`dev` -> `master`) was just done. Please close this issue if it is solved! Thanks.

closed 2020-04-29 by alonamid