ucb-bar/chipyard #203: Unable to complete building RISC-V Linux GNU Toolchain

hitoswal, 2019-08-16:
I am running the script to build riscv-tools. The basic tools build successfully.
When it builds on further it gets stuck at the command and exits
`mkdir -p stamps/ && touch stamps/build-binutils-linux`
which is part of building RISC-V Linux I believe, initiated from the line number 131.

I don't get any error messages but it doesn't print 
`\\nRISC-V Linux GNU Toolchain installation completed!` either. Neither it genetares env.sh.

On my further investigation, I found that it doesn't beyond line number 169 of Makefile in 
`<ChipyardRepo>/toolschains/riscv-gnu-toolchain/build/`

hitoswal, 2019-08-19:
```
make  install-recursive
make[10]: Entering directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb/build-gnulib/import'
make[11]: Entering directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb/build-gnulib/import'
make[12]: Entering directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb/build-gnulib/import'
if test yes = no; then \
  case 'linux-gnu' in \
    darwin[56]*) \
      need_charset_alias=true ;; \
    darwin* | cygwin* | mingw* | pw32* | cegcc*) \
      need_charset_alias=false ;; \
    *) \
      need_charset_alias=true ;; \
  esac ; \
else \
  need_charset_alias=false ; \
fi ; \
if $need_charset_alias; then \
  /bin/bash {path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/gnulib/../../mkinstalldirs {path_to_chipyard}/riscv-tools-install/lib ; \
fi ; \
if test -f {path_to_chipyard}/riscv-tools-install/lib/charset.alias; then \
  sed -f ref-add.sed {path_to_chipyard}/riscv-tools-install/lib/charset.alias > {path_to_chipyard}/riscv-tools-install/lib/charset.tmp ; \
  /usr/bin/install -c -m 644 {path_to_chipyard}/riscv-tools-install/lib/charset.tmp {path_to_chipyard}/riscv-tools-install/lib/charset.alias ; \
  rm -f {path_to_chipyard}/riscv-tools-install/lib/charset.tmp ; \
else \
  if $need_charset_alias; then \
    sed -f ref-add.sed charset.alias > {path_to_chipyard}/riscv-tools-install/lib/charset.tmp ; \
    /usr/bin/install -c -m 644 {path_to_chipyard}/riscv-tools-install/lib/charset.tmp {path_to_chipyard}/riscv-tools-install/lib/charset.alias ; \
    rm -f {path_to_chipyard}/riscv-tools-install/lib/charset.tmp ; \
  fi ; \
fi
make[12]: Nothing to be done for 'install-data-am'.
make[12]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb/build-gnulib/import'
make[11]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb/build-gnulib/import'
make[10]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb/build-gnulib/import'
make[9]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb/build-gnulib/import'
make[8]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb/build-gnulib'
make[7]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb/build-gnulib'
make[6]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb/build-gnulib'
make[5]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb'
make[4]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb'
make[3]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux/gdb'
make[2]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux'
make[1]: Leaving directory '{path_to_chipyard}/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-linux'
```
Above are the few lines of output before termination of the process.

colinschmidt, 2019-08-21:
This log does not show any errors? Perhaps you can use gist.github.com to show the full log which presumably includes errors.

hitoswal, 2019-08-21:
Here is the [log file](https://gist.github.com/hitoswal/45057f12572e02b8761b613dff4ad3ed)
I couldn't find any error.

zhemao, 2019-09-02:
That's quite odd. Can you tell us what system you are trying to install the toolchain on?

Can you try going into `toolchains/riscv-tools/riscv-gnu-toolchain/build` and manually running `make clean && make`?

zhemao, 2019-09-16:
Hmm, I guess there might be some GNU make race condition. If you run `make -j16 linux` manually, does it give you any error message?

We've never tested on OpenSUSE before, so thank you for bringing this to our attention.

andrewzigerelli, 2019-10-22:
I am on Ubuntu 16.04, I had this bug. I found that libmpfr version check was failing by doing a single threaded make manually.

apt install libmpfr-dev (or libmpc-dev) fixes it 

I actually apt install libmpc-dev, and apt install pulls in "recommended" libmpfr-dev, even though it doesn't seem to recommend it according to the online docs.

Maybe the bignum, multiprecision, etc libraries need to be added to requirements:
libmpfr-dev, libmpc-dev, libgmp-dev.

abejgonzalez, 2020-08-17:
Closing for now since there is no update on this.

closed 2020-08-17 by abejgonzalez

hmwill, 2021-02-06:
I ran into the same problem as reported here, that is, the build exits at 
`mkdir -p stamps/ && touch stamps/build-binutils-linux`

This is in the Linux environment on a Chromebook, which is Debian 10. To the comment from @andrewzigerelli above:

`apt install libmpfr-dev` did not work for me, but `apt install libmpc-dev` (as you seemed to have done) got me past this point.

patricksheu, 2024-09-24:
I encountered the same issue, but what’s strange is that the [riscv-gnu-toolchain GitHub repository](https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file) recommends using `libmpc` with Homebrew, and I'm already using it.

I'm on macOS 15, and I’ve tried both GCC 14 and Clang. Additionally, I'm working on a case-sensitive disk volume.

Here’s the process I’ve followed:

1. Installed necessary dependencies:
   ```bash
   brew install python3 gawk gnu-sed gmp mpfr libmpc isl zlib expat texinfo flock libslirp
   ```
2. Configured the toolchain:
   ```bash
   ./configure --prefix=/Users/patrick/riscv
   ```
3. Ran the make command:
   ```bash
   make
   ```

I’ve tested installing 
in `/opt/riscv`, /Users/patrick/riscv and `/Volumes/1T_SSD/source/tools/riscv`.

I’ve also tried using `sudo make`, `make -j8`, and modified various environment variables. I deleted the files and tried recompiling from scratch, updating the submodules multiple times.

Here’s my modified configuration command:
```bash
../configure --prefix=/Users/patrick/riscv \
--with-binutils=/opt/homebrew/opt/binutils \
--with-gcc=/opt/homebrew/opt/gcc \
--with-gdb=/opt/homebrew/opt/gdb \
--with-glibc=/opt/homebrew/opt/glibc \
--with-linux-headers=/opt/homebrew/opt/linux-headers \
--with-llvm=/opt/homebrew/opt/llvm \
--with-pk=/opt/homebrew/opt/pk \
--with-qemu=/opt/homebrew/opt/qemu
```

This is my log
[build.log](https://github.com/user-attachments/files/17114564/build.log)