ucb-bar/chipyard #232: Verilator fatal error: fesvr/dtm.h when trying to compile default simulator config

smuchand, 2019-09-08:
Hello, I was trying to build defaultRocketConfig and I am seeing this issue: 
**********************************************************
<pathtorepo>/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/SimDTM.cc:3:10: fatal error: fesvr/dtm.h: No such file or directory
 #include <fesvr/dtm.h>
          ^~~~~~~~~~~~~
compilation terminated.
VTestHarness.mk:67: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory '<pathtorepo>/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig'
Makefile:100: recipe for target '<pathtorepo>/sims/verilator/simulator-example-DefaultRocketConfig' failed
make: *** [<pathtorepo>/sims/verilator/simulator-example-DefaultRocketConfig] Error 2
***************************************************************

Thanks,
Shruthi

smuchand, 2019-09-08:
Sorry, my mistake I had my env variables unset.

closed 2019-09-08 by smuchand

HaFred, 2020-02-01:
Hi Shruthi, did you solve this issue?
I have my env vars set, but it still failed.

smuchand, 2020-02-01:
Hey, yes that was solved. Try `make clean` and delete the
sims/verilator/verilator_install and `make` again. If that doesn't work,
try to reclone the repo if possible and rebuild the toolchain and
everything.

Thanks.
Shruthi

ᐧ