ucb-bar/chipyard #233: sims/verilator build fails

Ravenwater, 2019-09-08:
Trying to build the Verilator simulation environment in a freshly cloned repo, and run into the following error:
```
touch /home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig/VTestHarness.mk
make VM_PARALLEL_BUILDS=1 -C /home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig -f VTestHarness.mk
make[1]: Entering directory '/home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig'
g++  -I.  -MMD -I/home/theodore/dev/clones/chipyard/sims/verilator/verilator_install/install/share/verilator/include -I/home/theodore/dev/clones/chipyard/sims/verilator/verilator_install/install/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O1 -std=c++11 -I/include -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -I/home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig -include /home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig.plusArgs -include /home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig/VTestHarness.h   -c -o SimDTM.o /home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/SimDTM.cc
/home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/SimDTM.cc:3:10: fatal error: fesvr/dtm.h: No such file or directory
 #include <fesvr/dtm.h>
          ^~~~~~~~~~~~~
compilation terminated.
VTestHarness.mk:67: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory '/home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig'
Makefile:100: recipe for target '/home/theodore/dev/clones/chipyard/sims/verilator/simulator-example-DefaultRocketConfig' failed
make: *** [/home/theodore/dev/clones/chipyard/sims/verilator/simulator-example-DefaultRocketConfig] Error 2
```

This was the sequence of events that lead to this:
```
$  git clone https://github.com/ucb-bar/chipyard
$  cd chipyard/
$  ./scripts/init-submodules-no-riscv-tools.sh 
$  ./scripts/build-toolchains.sh esp-tools
$  cd sims/verilator/
$  make
```
Is there a step that I missed to have fesvr/dtm.h in my environment?

davidbiancolin, 2019-09-08:
You probably need to source the `env.sh` generated by build-toolchains.sh. This puts the RISC-V toolchain and associated libraries on your path. This should be at the root of your chipyard repo.

Ravenwater, 2019-09-08:
that yields:

```
theodore@sw-vm-300:~/dev/clones/chipyard$ ls
bootrom    common.mk  env.sh             generators  LICENSE  README.md  sims   toolchains  variables.mk
build.sbt  docs       esp-tools-install  lib         project  scripts    tests  tools       vlsi
theodore@sw-vm-300:~/dev/clones/chipyard$ source env.sh
theodore@sw-vm-300:~/dev/clones/chipyard$ cd sims/verilator/
theodore@sw-vm-300:~/dev/clones/chipyard/sims/verilator$ make
make VM_PARALLEL_BUILDS=1 -C /home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig -f VTestHarness.mk
make[1]: Entering directory '/home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig'
g++  -I.  -MMD -I/home/theodore/dev/clones/chipyard/sims/verilator/verilator_install/install/share/verilator/include -I/home/theodore/dev/clones/chipyard/sims/verilator/verilator_install/install/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O1 -std=c++11 -I/include -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -I/home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig -include /home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig.plusArgs -include /home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig/VTestHarness.h   -c -o SimDTM.o /home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/SimDTM.cc
/home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/SimDTM.cc:3:10: fatal error: fesvr/dtm.h: No such file or directory
 #include <fesvr/dtm.h>
          ^~~~~~~~~~~~~
compilation terminated.
VTestHarness.mk:67: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory '/home/theodore/dev/clones/chipyard/sims/verilator/generated-src/example.TestHarness.DefaultRocketConfig/example.TestHarness.DefaultRocketConfig'
Makefile:100: recipe for target '/home/theodore/dev/clones/chipyard/sims/verilator/simulator-example-DefaultRocketConfig' failed
make: *** [/home/theodore/dev/clones/chipyard/sims/verilator/simulator-example-DefaultRocketConfig] Error 2
theodore@sw-vm-300:~/dev/clones/chipyard/sims/verilator$ find ../.. -name dtm.h
../../esp-tools-install/include/fesvr/dtm.h
../../toolchains/esp-tools/riscv-fesvr/fesvr/dtm.h
theodore@sw-vm-300:~/dev/clones/chipyard/sims/verilator$ cat ../../env.sh 
export CHIPYARD_TOOLCHAIN_SOURCED=1
export RISCV=/home/theodore/dev/clones/chipyard/esp-tools-install
export PATH=/home/theodore/dev/clones/chipyard/esp-tools-install/bin:/home/theodore/dev/clones/chipyard/:$PATH
export LD_LIBRARY_PATH=/home/theodore/dev/clones/chipyard/esp-tools-install/lib${LD_LIBRARY_PATH:+:/home/theodore/dev/clones/chipyard/esp-tools-install/lib}
theodore@sw-vm-300:~/dev/clones/chipyard/sims/verilator$ 
```

looking at the compile line, it does no appear to reference .../esp-tools-install/include that contains that library include

davidbiancolin, 2019-09-08:
Could you try cleaning out your verilator directory and rerunning? Verilator produces a bunch of make fragments that may be out-of-date now that you've set $RISCV.

Ravenwater, 2019-09-09:
Yep, that cleared up the build failure. I now see the errors of my ways: order needs to be followed!

Thank you for getting me over this hurdle.

closed 2019-09-09 by Ravenwater