ucb-bar/chipyard #299: VLSI tutorial

hqjenny, 2019-10-12:
https://github.com/ucb-bar/chipyard/blob/master/docs/VLSI/Tutorial.rst
1. There are extra markdown symbols in some commands.
e.g. \`\`./scripts/init-vlsi.sh asap7\`\`. 

2. In addition to `make syn`,  for the ASAP7 example without memory compiler, maybe add explicit instr like`make syn MACROCOMPILER_MODE='--mode synflops'`, so the synthesis won't fail with the error `hammer-vlsi/technology/asap7/sram-cache.json (No such file or directory)`.  It is unclear if the env var `MACROCOMPILER_MODE` needs to be explicitly passed to `make syn` again after `make buildfile`

abejgonzalez, 2019-10-13:
@harrisonliew @dpgrubb13 Could you guys look into this?

colinschmidt, 2019-10-16:
I have a fix for the first issue and am testing the second part to see if we need a change.

PR #312 (2019-10-16): [docs][ci skip] Remove extra backticks in docs
Also fix sram-cache error. Fixes #299
commit 701129fb (2019-10-16): [docs][ci skip] Remove extra backticks in docs (#312)

Closes #299
commit 784bff26 (2019-10-16): [docs][ci skip] Remove extra backticks in docs

Also fix sram-cache error. Fixes #299
closed 2019-10-16 by harrisonliew