ucb-bar/chipyard #300: Verilator build failure for Sha3RocketConfig

miniskar, 2019-10-13:
I am facing below error while doing the make for verilator. I have followed the build steps as given in the below document.
https://chipyard.readthedocs.io/en/latest/

Here is the make command and some error log. 

$ make CONFIG=Sha3RocketConfig -j16
[info] Packaging /home/nqx/RISCV/chipyard.Oct13.2019/generators/hwacha/target/scala-2.12/hwacha_2.
12-1.2.jar ...
[info] Done packaging.
[info] Compiling 12 Scala sources to /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/target
/scala-2.12/classes ...
[error] /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/src/main/scala/ConfigMixins.scala:1
8:22: object devices is not a member of package sifive.blocks
[error] import sifive.blocks.devices.gpio._
[error]                      ^
[error] /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/src/main/scala/ConfigMixins.scala:4
5:8: not found: value PeripheryGPIOKey
[error]   case PeripheryGPIOKey => List(
[error]        ^
[error] /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/src/main/scala/ConfigMixins.scala:4
6:5: not found: value GPIOParams
[error]     GPIOParams(address = 0x10012000, width = 4, includeIOF = false))
[error]     ^
[error] /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/src/main/scala/ConfigMixins.scala:4
6:16: not found: value address
[error]     GPIOParams(address = 0x10012000, width = 4, includeIOF = false))
[error]                ^
[error] /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/src/main/scala/ConfigMixins.scala:4
6:38: not found: value width
[error]     GPIOParams(address = 0x10012000, width = 4, includeIOF = false))
[error]                                      ^
[error] /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/src/main/scala/ConfigMixins.scala:4
6:49: not found: value includeIOF
[error]     GPIOParams(address = 0x10012000, width = 4, includeIOF = false))
[error]                                                 ^
[error] /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/src/main/scala/Top.scala:15:22: obj
ect devices is not a member of package sifive.blocks
[error] import sifive.blocks.devices.gpio._
[error]                      ^
[error] /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/src/main/scala/Top.scala:77:8: not 
found: type HasPeripheryGPIO
[error]   with HasPeripheryGPIO {
[error]        ^
[error] /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/src/main/scala/Top.scala:83:8: not 
found: type HasPeripheryGPIOModuleImp
[error]   with HasPeripheryGPIOModuleImp
[error]        ^
[error] /home/nqx/RISCV/chipyard.Oct13.2019/generators/example/src/main/scala/ConfigMixins.scala:1
27:22: value gpio is not a member of example.TopWithGPIOModule
[error]     for (gpio <- top.gpio) {
[error]                      ^
[error] 10 errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 43 s, completed Oct 13, 2019 12:30:01 AM
/home/nqx/RISCV/chipyard.Oct13.2019/common.mk:44: recipe for target 'generator_temp' failed
make: *** [generator_temp] Error 1

Getting this error for other configs also. Please help in resolving this issue. 

With Regards,
Narasinga.

abejgonzalez, 2019-10-13:
Did you run `./scripts/init-submodules-no-riscv-tools.sh`? Did you follow the following page: https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html?

miniskar, 2019-10-13:
Yes. I have followed the steps as per the instructions in that webpage.

abejgonzalez, 2019-10-13:
Can you show the output of `git status` and check that sources are there? What were the steps that you ran to get things started? Thanks!

abejgonzalez, 2019-10-13:
Sorry also could you show `git submodule status`

miniskar, 2019-10-13:
Step 1:
$ git clone https://github.com/ucb-bar/chipyard.git
$ cd chipyard
$ ./scripts/init-submodules-no-riscv-tools.sh

Step 2:
$ export MAKEFLAGS=-j8
$./scripts/build-toolchains.sh
# Successful 

Step 3:
$ source ./env.sh

Step 4:
$ cd sims/verilator
$ make CONFIG=Sha3RocketConfig -j16
# Tried other Rocket configurations and Boom configurations also. Getting same error. 

Here is the git status. 
nqx@amundsen:[02:05 PM]~/RISCV/chipyard.Oct13.2019$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

	modified:   generators/sifive-blocks (modified content, untracked content)
	modified:   generators/sifive-cache (untracked content)
	modified:   toolchains/esp-tools/riscv-gnu-toolchain (modified content)
	modified:   toolchains/qemu (untracked content)
	modified:   toolchains/riscv-tools/riscv-gnu-toolchain (modified content)
	modified:   tools/barstools (modified content)
	modified:   tools/torture (modified content)
	modified:   vlsi/hammer (modified content)

Here is the submodule status.
nqx@amundsen:[02:08 PM]~/RISCV/chipyard.bare.original$ git submodule status
 2a0ea2e7acfd4605eed513e15062848e4e5be309 generators/boom (2.2.2-260-g2a0ea2e7)
 ff4605f5d10d91b7ae8c4a006b965d4706009a06 generators/hwacha (remotes/origin/priv19-merge-104-gff4605f)
 baa40ed85d7425ef5ce206d52fb8b2759c6f6827 generators/icenet (heads/master-1-gbaa40ed)
 50de8a34c19c12de5066cd7ada50ebb5f5b2ea26 generators/rocket-chip (v1.2-073119-SNAPSHOT^2~21)
 60ddfe7c5b2c1640d91e7d9a35c65011c1600810 generators/sha3 (remotes/origin/jerryz123-patch-1-12-g60ddfe7)
 24dd537894379dc160ed9e15d33444439822ab5b generators/sifive-blocks (v1.0)
 13d0c2f17853a658ae86eae793718c71ac82dddf generators/sifive-cache (13d0c2f)
 aa13f6ccc1a05a20e52a1600b6c8c796d306f1cd generators/testchipip (remotes/origin/dev)
 7f36976cde51309834851ff4f246a2fe007ce324 sims/firesim (1.6.0-346-g7f36976)
 9f532293985d08b0c176d96c7b650e5f433780e1 toolchains/esp-tools/riscv-gnu-toolchain ((null))
 0ffa02e5b4ca57ec44684119a1a9a31b3871857b toolchains/esp-tools/riscv-isa-sim ((null))
 a3e4ac61d2b1ff37a22b9193b85d3b94273e80cb toolchains/esp-tools/riscv-pk ((null))
 f1370d054389fc83974fc820985b5c51693b8f9d toolchains/esp-tools/riscv-tests ((null))
 4f59102571fce49af180cfc6d4cdd2b5df7bdb14 toolchains/qemu ((null))
 2855d823a6e93d50af604264b02ced951e80de67 toolchains/riscv-tools/riscv-gnu-toolchain ((null))
 5e32a0157f91ebfb5c7ea7113fce28bf40016fa4 toolchains/riscv-tools/riscv-gnu-toolchain-prebuilt ((null))
 9443c1dbac0301faf3a47c5e6914cc7dcb34983e toolchains/riscv-tools/riscv-isa-sim ((null))
 7c82a7b9d5b7d8b71e0a66826705ec141db718c3 toolchains/riscv-tools/riscv-openocd ((null))
 a3e4ac61d2b1ff37a22b9193b85d3b94273e80cb toolchains/riscv-tools/riscv-pk ((null))
 249796cec94d75ff10ca034153e206a319e87158 toolchains/riscv-tools/riscv-tests ((null))
 26096e07f6ce3e12b2114132c2859ef56fb0cfaf tools/barstools (heads/master)
 41f4eef0d85b65fabd0d786efa8baa099513dcf0 tools/chisel-testers (2019-03-11-SNAPSHOT-7-g41f4eef)
 e1aa5f3f5c0cdeb204047c3ca50801d9f7ea25f1 tools/chisel3 (2019-03-11-SNAPSHOT-48-ge1aa5f3f)
 15145ab6230f869676de7eb730b4267fff7b11e8 tools/dsptools (2019-03-11-SNAPSHOT-31-g15145ab)
 84a1c7b1f7311ce036cb7d3d5eb652466b87dce4 tools/firrtl (2019-03-11-SNAPSHOT-61-g84a1c7b1)
 a881c07df6bceea462dbbd9a28e25721a1e88567 tools/firrtl-interpreter (2019-03-11-SNAPSHOT-6-ga881c07)
 59b0f0f224ff4f1eb6ebb1b4dd7eaf1ab3fac2e5 tools/torture (heads/master)
 a03b969af104770a7662ecdbac80cebf622e674b tools/treadle (2019-03-11-SNAPSHOT-17-ga03b969)
 85c12e98e63d599f5267c7711a24c174d74974c5 vlsi/hammer (heads/master)
 5f5d9d9e574d54acd3a84d1885c9f9b2897f373b vlsi/hammer-cadence-plugins ((null))
 33ccdccf2c04a26cceeeb03a29b9cfad38908328 vlsi/hammer-mentor-plugins ((null))
 e0ace7345e98e11b17ce550550c902782010e032 vlsi/hammer-synopsys-plugins ((null))

abejgonzalez, 2019-10-13:
Hmm. For some reason, your submodules are messed up. I just cloned Chipyard `master` in my own environment (ubuntu) and it worked for me. I would recommend doing a fresh clone and make sure that all the steps finish correctly (or you can try to fix all the submodules individually). In your case make sure the `init-submodules-no-riscv-tools.sh` step completes correctly. After running that command you should see the following in `git status`:

```
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean

$ git submodule status
2a0ea2e7acfd4605eed513e15062848e4e5be309 generators/boom (2.2.2-260-g2a0ea2e)
 ff4605f5d10d91b7ae8c4a006b965d4706009a06 generators/hwacha (remotes/origin/priv19-merge-104-gff4605f)
 baa40ed85d7425ef5ce206d52fb8b2759c6f6827 generators/icenet (heads/master-1-gbaa40ed)
 50de8a34c19c12de5066cd7ada50ebb5f5b2ea26 generators/rocket-chip (v1.2-073119-SNAPSHOT^2~21)
 60ddfe7c5b2c1640d91e7d9a35c65011c1600810 generators/sha3 (remotes/origin/jerryz123-patch-1-12-g60ddfe7)
 24dd537894379dc160ed9e15d33444439822ab5b generators/sifive-blocks (v1.0)
 13d0c2f17853a658ae86eae793718c71ac82dddf generators/sifive-cache (13d0c2f)
 aa13f6ccc1a05a20e52a1600b6c8c796d306f1cd generators/testchipip (remotes/origin/dev)
 7f36976cde51309834851ff4f246a2fe007ce324 sims/firesim (1.6.0-346-g7f36976)
-9f532293985d08b0c176d96c7b650e5f433780e1 toolchains/esp-tools/riscv-gnu-toolchain
-0ffa02e5b4ca57ec44684119a1a9a31b3871857b toolchains/esp-tools/riscv-isa-sim
-a3e4ac61d2b1ff37a22b9193b85d3b94273e80cb toolchains/esp-tools/riscv-pk
-f1370d054389fc83974fc820985b5c51693b8f9d toolchains/esp-tools/riscv-tests
-4f59102571fce49af180cfc6d4cdd2b5df7bdb14 toolchains/qemu
-2855d823a6e93d50af604264b02ced951e80de67 toolchains/riscv-tools/riscv-gnu-toolchain
-5e32a0157f91ebfb5c7ea7113fce28bf40016fa4 toolchains/riscv-tools/riscv-gnu-toolchain-prebuilt
-9443c1dbac0301faf3a47c5e6914cc7dcb34983e toolchains/riscv-tools/riscv-isa-sim
-7c82a7b9d5b7d8b71e0a66826705ec141db718c3 toolchains/riscv-tools/riscv-openocd
-a3e4ac61d2b1ff37a22b9193b85d3b94273e80cb toolchains/riscv-tools/riscv-pk
-249796cec94d75ff10ca034153e206a319e87158 toolchains/riscv-tools/riscv-tests
 26096e07f6ce3e12b2114132c2859ef56fb0cfaf tools/barstools (heads/master)
 41f4eef0d85b65fabd0d786efa8baa099513dcf0 tools/chisel-testers (2019-03-11-SNAPSHOT-7-g41f4eef)
 e1aa5f3f5c0cdeb204047c3ca50801d9f7ea25f1 tools/chisel3 (2019-03-11-SNAPSHOT-48-ge1aa5f3)
 15145ab6230f869676de7eb730b4267fff7b11e8 tools/dsptools (2019-03-11-SNAPSHOT-31-g15145ab)
 84a1c7b1f7311ce036cb7d3d5eb652466b87dce4 tools/firrtl (2019-03-11-SNAPSHOT-61-g84a1c7b)
 a881c07df6bceea462dbbd9a28e25721a1e88567 tools/firrtl-interpreter (2019-03-11-SNAPSHOT-6-ga881c07)
 59b0f0f224ff4f1eb6ebb1b4dd7eaf1ab3fac2e5 tools/torture (heads/master)
 a03b969af104770a7662ecdbac80cebf622e674b tools/treadle (2019-03-11-SNAPSHOT-17-ga03b969)
 85c12e98e63d599f5267c7711a24c174d74974c5 vlsi/hammer (heads/master)
-5f5d9d9e574d54acd3a84d1885c9f9b2897f373b vlsi/hammer-cadence-plugins
-33ccdccf2c04a26cceeeb03a29b9cfad38908328 vlsi/hammer-mentor-plugins
-e0ace7345e98e11b17ce550550c902782010e032 vlsi/hammer-synopsys-plugins
```

miniskar, 2019-10-14:
Yes. You are right. The checkout didn't happen properly. As my organization network do not support git:// protocol which is used to checkout the generators/rocket-chip/torture/env submodule, we have tweaked it to use https:// manually. Afterwards, the $git submodule sync made the messup it seems. 
I found another workaround to overcome this, which is working now.
Thank you for your help in addressing the issue. I am closing this ticket now.

closed 2019-10-14 by miniskar