ucb-bar/chipyard #310: fatal repository not found

nicolast0604, 2019-10-16:
Username for 'https://github.com': xxxx
Password for 'https://xxxx@github.com':
remote: Repository not found.
fatal: repository 'https://github.com/ucb-bar/hammer-cadence-plugins.git/' not found
fatal: clone of 'https://github.com/ucb-bar/hammer-cadence-plugins.git' into submodule path 'vlsi/hammer-cadence-plugins' failed

jerryz123, 2019-10-16:
How are you initializing submodules? 
Are you running `scripts/init-submodules-no-riscv-tools.sh`?

cousteaulecommandant, 2020-04-03:
Why have the project in .gitmodules at all if it no longer exists, though?
Also, although the documentation in https://chipyard.readthedocs.io/ already covers this, I think it would be a good idea to add a note at the top of README.md warning that users shouldn't clone this repository recursively.

alonamid, 2020-04-28:
The project still exists, but it is not public due to EDA user license agreements.
(More details about that can be found here: https://hammer-vlsi.readthedocs.io/en/latest/Hammer-Basics/Hammer-Overview.html#tool-plugins).
We would be happy to accept contributions for Hammer plug-ins for a fully-open source EDA stack. 

Chipyard is made up of many submodules, and if you were to clone it recursively, you would likely also clone the full Linux kernel 3 times times. There is a reason why Chipyard has a setup script, and that is how setup is instructed in the documentation.
We'll look into adding it to the README as well.

closed 2020-04-28 by alonamid