ucb-bar/chipyard #311: init submodules script still pulls in a lot of stuff non-firesim developers don't need

colinschmidt, 2019-10-16:
Perhaps we can have a `scripts/init-submodules-no-firesim.sh`?

davidbiancolin, 2019-10-16:
Is it pulling more than FireSim and Midas?

colinschmidt, 2019-10-16:
Well riscv-linux is multi gigabyte so that really slows down the update.

jerryz123, 2019-10-16:
I feel like marshal should lazily init submodules when it needs to build it.

colinschmidt, 2019-10-16:
And I seem to get multiple copies of riscv-linux
`sw/firesim-software/riscv-linux`
and another one thats still going.

colinschmidt, 2019-10-16:
here was the other riscv-linx `sw/firesim-software/workloads/sha3/riscv-linux`

davidbiancolin, 2019-10-16:
To be clear, i was asking because the only submodule in FireSim that should be initialized by default is Midas -- if that's not the case i'll fix it.

a0u, 2019-10-17:
`scripts/init-submodules-no-riscv-tools.sh --no-firesim` can be used to skip the firesim initialization entirely, including FireMarshal, but that probably breaks the Scala dependencies.

FireMarshal is being migrated from firesim to chipyard, which is why it was added to the initialization, but it could also be deferred until `build-toolchains.sh` or later.

davidbiancolin, 2019-10-18:
Yeah i feel like the right solution is to push marshal initialization into a later script. Previously, init-submodules-no-riscv-tools.sh just initialized firesim and _only_ the midas submodule, which is basically a NOP in the initialization process but is required to compile scala sources currently.

I'm surprised --no-firesim works.

alonamid, 2020-10-19:
Fixed a while ago (there is now a separate firesim-setup script and a firemarshal submodule initialization script)

closed 2020-10-19 by alonamid