ucb-bar/chipyard #323: Verilog generation speed?

austinharris, 2019-11-01:
I realize there is quite a bit more going on in chipyard compared to generating the verilog from the vsim directory of rocket-chip, but I am still quite surprised at how slow it is just to generate verilog to use on my (non-firesim) FPGA. Will it make much difference to remove some of the projects I am not using (e.g., icenet, firesim, midas)? Or perhaps using the old vlsi_mem_gen flow instead of the barstools macro compiler?

Are there some other opportunities to speed up SBT as in #74 or any other low-hanging fruit to speed things up?

Thanks!

colinschmidt, 2019-11-01:
We should first figure out the breakdown of time for a chipyard verilog generation vs rocket-chip verilog generation.

That being said I don't think barstools macro compiler is a significant contributor. For a simple config its <2% of my run time. Depending on what you are doing avoiding generating the separate harness file could be low-hanging fruit, but that is only useful in a small set of use cases. Potentially updating barstools to "fork" the circuit into test harness and dut in a single run of firrtl would speed things up by a good amount.

austinharris, 2019-11-01:
This isn't a direct comparison, but my previous dual-core rocket looks like this:
- 42s for the initial firrtl elaboration
- 36s for verilog generation

Compared to my medium single boom in chipyard:
- 108s for initial firrtl elaboration
- 830s for GenerateTopAndHarness
- 14s and 11s for macros

Indeed it doesn't seem worth it to change the macro compiler. I don't believe these times include initializing sbt which is quite a bit longer here as well.

jerryz123, 2019-11-01:
The `MediumBoom` times match what I experience.

colinschmidt, 2019-11-01:
Can you record numbers for a dual-core rocket in chipyard?

For Boom you should add `-ll info` to the firrtl run to see if any particular pass looks like an outlier.

colinschmidt, 2019-11-01:
Another arbitrary point this is for the default config in both (but I'm on a dev branch in chipyard so it is probably slower than normal) based on the reported times from each of the steps.
```
Chipyard
Elaboration | 29.587
Firrtl Top | 73.541
Firrtl Harness | 20.754
MacroCompiler | 2.363
```
```
Rocket-chip
Elaboration | 36.77
Firrtl Top | 59.106

```

austinharris, 2019-11-13:
We can close this if preferred since it looks like the majority of the difference is just boom vs. rocket.

Since the sbt startup is quite long for me it did help a bit to combine the 3 barstools commands into one sbt invocation separated with semicolons. I didn't look into the makefile much though so perhaps there are some settings where it needs to be separated.

I did notice that the newer wake stuff in rocket-chip seems to be even quicker, I guess it uses bloop instead?

closed 2020-04-29 by alonamid