ucb-bar/chipyard #324: unable to clone "wlutil/busybox"

icessspark, 2019-11-04:
To initialize the  repository setup, I followed the Docs and executed the follow instructions:
-----------------------------------------------------------------
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
./scripts/init-submodules-no-riscv-tools.sh
-----------------------------------------------------------------

And get the errors below:
------------------------------------------------------------------
Cloning into 'wlutil/busybox'...
error: GnuTLS recv error (-9): A TLS packet with unexpected length was received. (curl_result = 56, http_code = 200, sha1 = fa24f0c9ff2a1ce8aad7bc176c4206431da59d0e)
error: Unable to find 6f8bc904287d2c845159ba147516f8b197d7e4d5 under https://git.busybox.net/busybox
Cannot obtain needed blob 6f8bc904287d2c845159ba147516f8b197d7e4d5
while processing commit 8b77a9ea81a0bd89ee69e7742d9b920dd1562763.
error: fetch failed.
fatal: clone of 'https://git.busybox.net/busybox/' into submodule path 'wlutil/busybox' failed
Failed to recurse into submodule path 'sw/firesim-software'
--------------------------------------------------------------------

I'm new to all these stuff. Is there anything wrong with the busybox branch?

colinschmidt, 2019-11-04:
This might have been a temporary error, can you retry the script?

abejgonzalez, 2019-11-04:
We also get a failure on this during CI. The quick fix there was to just rerun CI.

sagark, 2019-11-04:
we should switch this to a github mirror @NathanTP

ninhho, 2020-01-17:
Hi,

I am getting below error when source ./scripts/init-submodules-no-riscv-tools.sh
Initialized empty Git repository in /scratch/den_scratch/hninh/chipyard/sims/firesim/target-design/chipyard/sims/firesim/target-design/chipyard/sims/firesim/sw/firesim-software/wlutil/busybox/.git/
error: garbage at end of loose object '4ecfe000136468a88fb7b3e95f2682a08d86c014'
fatal: object 4ecfe000136468a88fb7b3e95f2682a08d86c014 is corrupted
Clone of 'https://git.busybox.net/busybox/' into submodule path 'wlutil/busybox' failed
Failed to recurse into submodule path 'sw/firesim-software'
Failed to recurse into submodule path 'sims/firesim'
Failed to recurse into submodule path 'target-design/chipyard'
Failed to recurse into submodule path 'sims/firesim'
Failed to recurse into submodule path 'target-design/chipyard'
Failed to recurse into submodule path 'sims/firesim'

Does anyone get the same issue with me? Please help me out of this. Thanks.

abejgonzalez, 2020-01-17:
Go ahead and rerun the `./scripts/init-submodules-no-riscv-tools.sh` script. Until we switch this to something like a GitHub mirror, the `busybox` submodule will occasionally fail to clone.

NathanTP, 2020-01-20:
Incidentally, this is fixed in FireMarshal's dev branch. It will trickle up to Chipyard the next time we bump.

ninhho, 2020-01-22:
Thanks @abejgonzalez.

closed 2020-04-28 by alonamid