ucb-bar/chipyard #325: Verilator make run-fast fails on Sha3RocketConfig

20dy3, 2019-11-04:
Hi,

We are students trying to run Verilator benchmarks for the Rocket Core with the Sha3 RoCC example accelerator. However, we're running into this issue:

[image]

We followed the documentation [here](https://chipyard.readthedocs.io/en/latest/) and [here](https://chipyard.readthedocs.io/en/latest/Simulation/Software-RTL-Simulation.html#). `make CONFIG=Sha3RocketConfig` seems to work correctly, and the benchmarks run successfully on the vanilla Rocket Core with `make run-fast`.

Thanks in advance!

abejgonzalez, 2019-11-12:
Hi,

On the `dev` branch of Chipyard, I just ran `make CONFIG=Sha3RocketConfig run-asm-tests` and `run-bmark-tests` and `run-fast` and all three passed. It looks like you are using the `esp-tools` toolchain. You should use the normal toolchain since `esp-tools` is mainly used for Hwacha. Otherwise, can you give more information on what your environment is?

20dy3, 2019-11-13:
Thanks, not using `esp-tools` seems to work.

closed 2019-11-13 by 20dy3