ucb-bar/chipyard #350: Fails generating GPIORocketConfig

mcarman8, 2019-11-25:
I'm trying to make a configuration with available GPIO pins, but I get the error below when I run the command `make verilog config=GPIORocketConfig` for the verilator simulation.

> [error] (run-main-0) java.lang.reflect.InvocationTargetException
[error] java.lang.reflect.InvocationTargetException
[error] 	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error] 	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error] 	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error] 	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities.$anonfun$elaborate$1(GeneratorUtils.scala:54)
[error] 	at chisel3.core.Module$.do_apply(Module.scala:52)
[error] 	at chisel3.Driver$.$anonfun$elaborate$1(Driver.scala:95)
[error] 	at chisel3.internal.Builder$.$anonfun$build$2(Builder.scala:343)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:341)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at chisel3.internal.Builder$.build(Builder.scala:341)
[error] 	at chisel3.Driver$.elaborate(Driver.scala:95)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities.elaborate(GeneratorUtils.scala:59)
[error] Caused by: java.lang.ClassCastException: class scala.runtime.BoxedUnit cannot be cast to class example.TopModule (scala.runtime.BoxedUnit is in unnamed module of loader java.net.URLClassLoader @11455813; example.TopModule is in unnamed module of loader sbt.internal.inc.classpath.ClasspathUtilities$$anon$1 @4a68dd61)
[error] 	at example.TestHarness.<init>(TestHarness.scala:31)
[error] 	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error] 	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error] 	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error] 	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities.$anonfun$elaborate$1(GeneratorUtils.scala:54)
[error] 	at chisel3.core.Module$.do_apply(Module.scala:52)
[error] 	at chisel3.Driver$.$anonfun$elaborate$1(Driver.scala:95)
[error] 	at chisel3.internal.Builder$.$anonfun$build$2(Builder.scala:343)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:341)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at chisel3.internal.Builder$.build(Builder.scala:341)
[error] 	at chisel3.Driver$.elaborate(Driver.scala:95)
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 21 s, completed Nov 25, 2019, 4:35:10 PM

Anyone know what might be the cause of this? Thank you to anyone who can help.

abejgonzalez, 2019-11-26:
Try this: `make verilog CONFIG=GPIORocketConfig TOP=TopWithGPIO`

mcarman8, 2019-11-26:
That gives the same error

abejgonzalez, 2019-11-26:
Hmm. In my clean Chipyard repository on `dev` branch, this seems to work for me. Did you make modifications to Chipyard? It looks like the unit at `example.TestHarness.(TestHarness.scala:31)` is giving an error.

mcarman8, 2019-12-03:
It seems that at some point a config I added caused it not to work. I did need to make sure to include the TOP=TopWithGPIO line for anyone in the future looking at this thread.

closed 2019-12-03 by mcarman8