ucb-bar/chipyard #351: chipyard/sims/verilator makefile bug?

dsw, 2019-11-26:
I work through an ssh connection, so in case the connection drops, I always run long jobs, such as a makefile, through nohup and in the background.

The instructions here https://chipyard.readthedocs.io/en/latest/Simulation/Software-RTL-Simulation.html#sw-rtl-sim-intro say to run make in chipyard/sims/verilator.

On my Ubuntu 14.04 machine, if I run the makefile under nohup in the background, it just goes into the stopped state and I see a stopped job and htop shows nothing happening.  However, if I move it to the foreground it runs to completion.  I have run it in the background several times and it never works.

I think makefiles should work when run in the background.  I have never seen a program produce this kind of behavior before.  Sounds like a bug in something, but I cannot even think of how it is possible so I cannot conjecture the bug; perhaps I do not know Linux/Unix well enough.

ucbjrl, 2019-11-26:
If it's stopping after launching `sbt`, it may be due to `sbt` trying to read from stdin. A solution is to cat /dev/null to stdin when you want to run the job in the background.

sequencer, 2019-11-26:
Give a try to tmux ;p

dsw, 2020-01-26:
Attempting to follow your recommendation to redirect /dev/null into
the build, in rocket-chip I run the below under nohup and in the
background (from within another makefile).  The result is that (1)
some processes run for a few seconds, which I can see in htop,
resulting in the output below (modulo renaming directories), (2) then,
as I reported previously, everything hangs, and I can see in htop that
no processes are running.  As soon as I move the build process from
the background to the foreground, suddenly in htop I can see all of
the processes running again and the build runs to completion.

It would really be great to have a solution to this.  I have tried
using grep to find where various things are happening, but these
autogenerated makefiles are so weird and there are so many layers of
scripts and makefiles and other things calling each other that I
cannot find where the make targets that being run are defined or the
various command lines I see in the nohup.out are even being requested
by makefiles or scripts.  Any help is appreciated.

____
make -C emulator all < /dev/null
make[1]: Entering directory '/home/dsw/rocket-chip/emulator'
mkdir -p /home/dsw/rocket-chip/emulator/generated-src/
cd /home/dsw/rocket-chip && java -Xmx2G -Xss8M
-XX:MaxPermSize=256M -jar /home/dsw/rocket-chip/sbt-launch.jar
"runMain freechips.rocketchip.system.Generator
/home/dsw/rocket-chip/emulator/generated-src
freechips.rocketchip.system TestHarness freechips.rocketchip.system
DefaultConfig"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M;
support was removed in 8.0
____

jwright6323, 2020-01-27:
> Give a try to tmux ;p

This or GNU `screen` are probably your best bets. Most of us use one or the other and are able to disconnect from SSH without any issues.

colinschmidt, 2020-01-27:
As Jim mentioned it seems to be related to sbt. Maybe something in this stackoverflow will help you?
https://stackoverflow.com/questions/40643208/how-to-run-sbt-as-daemon

dsw, 2020-01-27:
Ah, thank you.  The second part of one of the answers on that thread
(embedded below) says to use setsid.  I have to say that sbt is very poorly
designed if it admits of this failure mode.

Following the recommendation of that answer to use setsid seems to work.  I
run it this way:

(1) Write a Makefile in rocket-chip having the following target:

  .PHONY: all
  all:
       setsid ${MAKE} -C emulator $@

(2) From the command line, do this:

  $ nohup time nice make all &
  $ tail -f nohup.out

https://stackoverflow.com/questions/40643208/how-to-run-sbt-as-daemon

____
Short answer: run sbt as follows:

setsid nohup sbt run &

Rationale:

The reason why sbt stops is arrival of SIGTTOU signal. It is delivered to
background process in several cases, which include modifying terminal
configuration. This is our case because according to strace -f sbt run &,
sbt does a lot of black magic under the hood like this:

[pid 16600] execve("/usr/bin/sh", ["sh", "-c", "stty -g < /dev/tty"], [/*
75 vars */] <unfinished ...>

To work this around, you can run sbt in a different session to detach it
from current terminal, so that it won't open /dev/tty and mess with our
terminal.
____

jwright6323, 2020-01-31:
This seems to be resolved

closed 2020-01-31 by jwright6323