ucb-bar/chipyard #360: How to parameterize Rocket core and BOOM to support 64-bit bus address width?

huoxinglaizhe, 2019-12-08:
How to parameterize Rocket core and BOOM to support 64-bit bus address width?
Thanks!

colinschmidt, 2019-12-12:
TileLink and diplomacy will automatically expand the address widths as needed based on the system's memory map.
What are you trying to do with this larger address bit width?

huoxinglaizhe, 2019-12-16:
We are building a system with more than 4Gb memory.
We tried to modify one memory to 33-bit address width and got compile errors as below:

"[error] /ff-share/chipyard/generators/rocket-chip/src/main/scala/tilelink/SRAM.scala:257:51: integer number too large
[error]   val ram  = LazyModule(new TLRAM(AddressSet(0x0, 0x1ffffffff), beatBytes = ramBeatBytes))"

It seems that function AdressSet can not support more than 32-bit address width, we tried to fix that limitation and got more compile errors.
Is that the right way to modify system's memory map? If it is then how to update function AdressSet?

colinschmidt, 2019-12-16:
I think you just need to tell scala your large constant is of the long type by appending an `L`.

```
val ram = LazyModule(new TLRAM(AddressSet(0x0, 0x1ffffffffL), beatBytes = ramBeatBytes))
```

huoxinglaizhe, 2020-02-24:
Thanks very much.

closed 2020-02-24 by huoxinglaizhe