ucb-bar/chipyard #373: build-toolchains.sh fails to probe the correct GNU Make version

abdullahyildiz, 2020-01-05:
When I run

`./scripts/build-toolchains.sh
`

I get the following error:

`build-toolchains.sh: obsolete make version; need GNU make 4.x or later`

However, I have already installed GNU Make 4.1 in my machine:

```
make --version
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law. 
```

PS By the way, I noticed that `MAKE` variable has not been set in the script.

a0u, 2020-01-05:
`MAKE` is set when `build-toolchains.sh` sources `build-util.sh`.

https://github.com/ucb-bar/chipyard/blob/c7b252aa09773364781a03bbf498782573ee8585/scripts/build-util.sh#L17

Check if `gnumake` or `gmake` on your system is a different version.

abdullahyildiz, 2020-01-06:
@a0u,

There are no gnumake or gmake commands in my machine.

a0u, 2020-01-06:
Could you rerun `build-toolchains.sh` with `set -x` added to the beginning of the script (where the other `set` commands are grouped), and paste the debug log here?

abdullahyildiz, 2020-01-06:
@a0u,

```
++ pwd
+ RDIR=/eda/chipyard
++ git rev-parse --show-toplevel
+ CHIPYARD_DIR=/eda/chipyard
+ TOOLCHAIN=riscv-tools
+ EC2FASTINSTALL=false
+ getopts hH-: opt
+ shift 0
+ '[' '' = ec2fast ']'
+ '[' -n '' ']'
+ INSTALL_DIR=riscv-tools-install
++ pwd
+ RISCV=/eda/chipyard/riscv-tools-install
+ export RISCV=/eda/chipyard/riscv-tools-install
+ RISCV=/eda/chipyard/riscv-tools-install
+ cd /eda/chipyard
++ pwd
+ SRCDIR=/eda/chipyard/toolchains/riscv-tools
+ '[' -d /eda/chipyard/toolchains/riscv-tools ']'
+ . ./scripts/build-util.sh
++ '[' -n /eda/chipyard/toolchains/riscv-tools ']'
+++ getconf _NPROCESSORS_ONLN
++ ncpu=1
++ case ${ncpu} in
++ export 'MAKEFLAGS=-j 1 '
++ MAKEFLAGS='-j 1 '
+++ command -v gnumake
+++ command -v gmake
+++ command -v make
++ MAKE=/usr/bin/make
++ readonly MAKE
+ '[' false = true ']'
+ read -r makever
+ /usr/bin/make --version
+ case ${makever} in
+ die 'obsolete make version; need GNU make 4.x or later'
+ error 'obsolete make version; need GNU make 4.x or later'
+ echo 'build-toolchains.sh: obsolete make version; need GNU make 4.x or later'
build-toolchains.sh: obsolete make version; need GNU make 4.x or later
+ exit -1
```

a0u, 2020-01-06:
Could you try the following patch and let me know what `$makever` is set to?  Does `/usr/bin/make --version` correspond to 4.1 as expected?

```patch
diff --git a/scripts/build-toolchains.sh b/scripts/build-toolchains.sh
index b3195ad..78a258d 100755
--- a/scripts/build-toolchains.sh
+++ b/scripts/build-toolchains.sh
@@ -99,8 +99,9 @@ if [ "${EC2FASTINSTALL}" = true ] ; then
     git submodule deinit "${module}" || :
 
 else
-    "${MAKE}" --version | (
+    "${MAKE}" --version 2>&1 | (
         read -r makever
+        echo "make: ${makever}"
         case ${makever} in
         'GNU Make '[4-9]\.*|'GNU Make '[1-9][0-9]) ;;
         *) false ;;
```

On another note, it is a bit unusual to see `read -r makever` precede `/usr/bin/make --version` in your debug output, which is an inversion of the pipeline sequence.  In contrast, various Bash 4.x and 5.x versions in my testing environments consistently produce:
```
++ MAKE=/usr/bin/make
++ readonly MAKE
+ '[' false = true ']'
+ /usr/bin/make --version
+ read -r makever
+ case ${makever} in
```

If you could provide some more information about your system, it would help identify any differences.
* Bash version (`/usr/bin/env bash --version`)
* Your Linux distro release (`cat /etc/os-release` if exists)

Thanks for your cooperation.

abdullahyildiz, 2020-01-07:
@a0u,

Here is the information about my system:

```
$ /usr/bin/env bash --version
GNU bash, version 4.4.20(1)-release (x86_64-pc-linux-gnu)
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

```
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.3 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

a0u, 2020-01-07:
By any chance, do you have a locale set in your environment with a non-Latin character set?

I am sometimes able to reproduce the same behavior on an Ubuntu 18.04 instance this way:
```
$ LC_ALL='zh_CN.UTF-8' ./scripts/build-toolchains.sh
...
+ /usr/bin/make --version
+ read -r makever
+ echo 'make: GNU Make 4.1'
make: GNU Make 4.1
+ case ${makever} in
+ die 'obsolete make version; need GNU make 4.x or later'
+ error 'obsolete make version; need GNU make 4.x or later'
+ echo 'build-toolchains.sh: obsolete make version; need GNU make 4.x or later'
build-toolchains.sh: obsolete make version; need GNU make 4.x or later
+ exit -1
```
That might change the interpretation of the range expression (`[4-9]`) in the glob pattern and possibly account for this discrepancy.  Does it succeed if you set `LC_ALL=C` before the command as above?

abdullahyildiz, 2020-01-07:
@a0u,

```
$ LC_ALL='C' ./scripts/build-toolchains.sh
++ pwd
+ RDIR=/eda/chipyard
++ git rev-parse --show-toplevel
+ CHIPYARD_DIR=/eda/chipyard
+ TOOLCHAIN=riscv-tools
+ EC2FASTINSTALL=false
+ getopts hH-: opt
+ shift 0
+ '[' '' = ec2fast ']'
+ '[' -n '' ']'
+ INSTALL_DIR=riscv-tools-install
++ pwd
+ RISCV=/eda/chipyard/riscv-tools-install
+ export RISCV=/eda/chipyard/riscv-tools-install
+ RISCV=/eda/chipyard/riscv-tools-install
+ cd /eda/chipyard
++ pwd
+ SRCDIR=/eda/chipyard/toolchains/riscv-tools
+ '[' -d /eda/chipyard/toolchains/riscv-tools ']'
+ . ./scripts/build-util.sh
++ '[' -n /eda/chipyard/toolchains/riscv-tools ']'
+++ getconf _NPROCESSORS_ONLN
++ ncpu=1
++ case ${ncpu} in
++ export 'MAKEFLAGS=-j 1 '
++ MAKEFLAGS='-j 1 '
+++ command -v gnumake
+++ command -v gmake
+++ command -v make
++ MAKE=/usr/bin/make
++ readonly MAKE
+ '[' false = true ']'
+ /usr/bin/make --version
+ read -r makever
+ echo 'make: GNU Make 4.1'
make: GNU Make 4.1
+ case ${makever} in
+ die 'obsolete make version; need GNU make 4.x or later'
+ error 'obsolete make version; need GNU make 4.x or later'
+ echo 'build-toolchains.sh: obsolete make version; need GNU make 4.x or later'
build-toolchains.sh: obsolete make version; need GNU make 4.x or later
+ exit -1
```

a0u, 2020-01-08:
I think you'll have to investigate why the `case` pattern fails to match, given that the make version string was evidently read correctly.  All I can say is that a fresh 18.04 host with the same bash and make versions seems to have no issues for me under the default `en_US.UTF-8` locale.

Perhaps start with this change, and then try a few variations until it succeeds.
```patch
diff --git a/scripts/build-toolchains.sh b/scripts/build-toolchains.sh
index b3195ad..b8e35c7 100755
--- a/scripts/build-toolchains.sh
+++ b/scripts/build-toolchains.sh
@@ -102,7 +102,7 @@ else
     "${MAKE}" --version | (
         read -r makever
         case ${makever} in
-        'GNU Make '[4-9]\.*|'GNU Make '[1-9][0-9]) ;;
+        'GNU Make '[456789].*|'GNU Make '[123456789][0123456789]*) true ;;
         *) false ;;
         esac; ) || die 'obsolete make version; need GNU make 4.x or later'
```

abdullahyildiz, 2020-01-08:
@a0u, I give up :)

This the output of build-toolchains.sh:

```
++ pwd
+ RDIR=/eda/chipyard
++ git rev-parse --show-toplevel
+ CHIPYARD_DIR=/eda/chipyard
+ TOOLCHAIN=riscv-tools
+ EC2FASTINSTALL=false
+ getopts hH-: opt
+ shift 0
+ '[' '' = ec2fast ']'
+ '[' -n '' ']'
+ INSTALL_DIR=riscv-tools-install
++ pwd
+ RISCV=/eda/chipyard/riscv-tools-install
+ export RISCV=/eda/chipyard/riscv-tools-install
+ RISCV=/eda/chipyard/riscv-tools-install
+ cd /eda/chipyard
++ pwd
+ SRCDIR=/eda/chipyard/toolchains/riscv-tools
+ '[' -d /eda/chipyard/toolchains/riscv-tools ']'
+ . ./scripts/build-util.sh
++ '[' -n /eda/chipyard/toolchains/riscv-tools ']'
+++ getconf _NPROCESSORS_ONLN
++ ncpu=1
++ case ${ncpu} in
++ export 'MAKEFLAGS=-j 1 '
++ MAKEFLAGS='-j 1 '
+++ command -v gnumake
+++ command -v gmake
+++ command -v make
++ MAKE=/usr/bin/make
++ readonly MAKE
+ '[' false = true ']'
+ read -r makever
+ /usr/bin/make --version
+ case ${makever} in
+ true
+ die 'obsolete make version; need GNU make 4.x or later'
+ error 'obsolete make version; need GNU make 4.x or later'
+ echo 'build-toolchains.sh: obsolete make version; need GNU make 4.x or later'
build-toolchains.sh: obsolete make version; need GNU make 4.x or later
```
I attached the patched build-toolchains.sh.

[build-toolchains.sh.txt](https://github.com/ucb-bar/chipyard/files/4034191/build-toolchains.sh.txt)

dreamflyings, 2020-01-14:
@a0u is right. I am from China.In my environment , Command Line is still like this:  LC_ALL='zh_CN.UTF-8' ./build-toolchains.sh

abdullahyildiz, 2020-01-14:
@dreamflyings How did you fix it? The solution proposed by @a0u didn't work for me.

liwei039, 2020-01-14:
@a0u 
I have same above issue, below is the log info:
bogon:chipyard liwei$ pwd
/Users/liwei/Documents/Projects/Chipyard/chipyard
bogon:chipyard liwei$ which make
/usr/local/bin/make
bogon:chipyard liwei$ make -v
GNU Make 4.2
Built for x86_64-apple-darwin19.2.0
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
bogon:chipyard liwei$ ./scripts/build-toolchains.sh
+ set -e
+ set -o pipefail
++ pwd
+ RDIR=/Users/liwei/Documents/Projects/Chipyard/chipyard
++ git rev-parse --show-toplevel
+ CHIPYARD_DIR=/Users/liwei/Documents/Projects/Chipyard/chipyard
+ TOOLCHAIN=riscv-tools
+ EC2FASTINSTALL=false
+ getopts hH-: opt
+ shift 0
+ '[' '' = ec2fast ']'
+ '[' -n '' ']'
+ INSTALL_DIR=riscv-tools-install
++ pwd
+ RISCV=/Users/liwei/Documents/Projects/Chipyard/chipyard/riscv-tools-install
+ export RISCV=/Users/liwei/Documents/Projects/Chipyard/chipyard/riscv-tools-install
+ RISCV=/Users/liwei/Documents/Projects/Chipyard/chipyard/riscv-tools-install
+ cd /Users/liwei/Documents/Projects/Chipyard/chipyard
++ pwd
+ SRCDIR=/Users/liwei/Documents/Projects/Chipyard/chipyard/toolchains/riscv-tools
+ '[' -d /Users/liwei/Documents/Projects/Chipyard/chipyard/toolchains/riscv-tools ']'
+ . ./scripts/build-util.sh
++ '[' -n /Users/liwei/Documents/Projects/Chipyard/chipyard/toolchains/riscv-tools ']'
+++ getconf _NPROCESSORS_ONLN
++ ncpu=8
++ case ${ncpu} in
++ export 'MAKEFLAGS=-j 8 '
++ MAKEFLAGS='-j 8 '
+++ command -v gnumake
++ MAKE=/usr/bin/gnumake
++ readonly MAKE
+ '[' false = true ']'
+ /usr/bin/gnumake --version
+ read -r makever
+ case ${makever} in
+ false
+ die 'obsolete make version; need GNU make 4.x or later'
+ error 'obsolete make version; need GNU make 4.x or later'
+ echo 'build-toolchains.sh: obsolete make version; need GNU make 4.x or later'
build-toolchains.sh: obsolete make version; need GNU make 4.x or later
+ exit -1

dreamflyings, 2020-01-14:
You first check the environment with this command: locale 
If the displayed result is like this "LANG=C.UTF-8" or other,  you can do the following. 
user$ LC_ALL='en_US.UTF-8' ./build-toolchains.sh

Or the order I gave earlier

liwei039, 2020-01-14:
> You first check the environment with this command: locale
> If the displayed result is like this "LANG=C.UTF-8" or other, you can do the following.
> user$ LC_ALL='en_US.UTF-8' ./build-toolchains.sh
> 
> Or the order I gave earlier
still same issue

$ LC_ALL='en_US.UTF-8' ./scripts/build-toolchains.sh
build-toolchains.sh: obsolete make version; need GNU make 4.x or later

a0u, 2020-01-14:
@liwei039: Yours is different issue.  `gnumake` and `gmake`, if found, are prioritized over `make`.  Your debug log shows that `/usr/local/bin/make` is not being used:
> `+ MAKE=/usr/bin/gnumake`

This may seem peculiar, but it follows the logic of glibc's autoconf check for make:
https://github.com/riscv/riscv-glibc/blob/4e2943456e690d89f48e6e710757dd09404b0c9a/configure#L4812-L4873

abdullahyildiz, 2020-01-15:
@dreamflyings,

I'm still getting the same error:

```
LC_ALL='en_US.UTF-8' &&  ./scripts/build-toolchains.sh
/usr/bin/make
build-toolchains.sh: obsolete make version; need GNU make 4.x or later
```

liwei039, 2020-01-15:
> @liwei039: Yours is different issue. `gnumake` and `gmake`, if found, are prioritized over `make`. Your debug log shows that `/usr/local/bin/make` is not being used:
>
>
> This may seem peculiar, but it [...]

@a0u so how to fix this issue? I use MacbookPro, which I don't have right to change in /usr/bin, is there any solution? thanks a lot!

abdullahyildiz, 2020-01-15:
@liwei039,

For now, you can try to install the toolchain by commenting out the related lines of the script and making the script continue. That was what I did.

liwei039, 2020-01-15:
> @liwei039,
> 
> For now, you can try to install the toolchain by commenting out the related lines of the script and making the script continue. That was what I did.

Thanks a lot!
Got it, you mean that commenting out the "make version check", right?
So, the env still works well in your local dir?

abdullahyildiz, 2020-01-15:
@liwei039,
> Got it, you mean that commenting out the "make version check", right?
> So, the env still works well in your local dir?

Yes, it worked for me.

siyy123, 2020-02-03:
I've encountered the same problem but sometimes if I do "make -v" before then it would work.

OmarDez, 2020-02-13:
It works but when updating autoconf files for riscv-gnu-toolchain says:
`autoreconf: 'configure.ac or 'configure.in' is required` what to do?

yunusdawji, 2020-05-12:
you have to a parameter called MAKE in build_util.sh

abejgonzalez, 2020-08-17:
Closing for now since this seems to be solved.

closed 2020-08-17 by abejgonzalez