ucb-bar/chipyard #376: Error for test Sha3-rocc and hwacha examples

dreamflyings, 2020-01-13:
hi, I have compiled rocket using :
make debug CONFIG=Sha3RocketConfig
toolchains is: chipyard-toolchain-prebuilt https://github.com/ucb-bar/chipyard-toolchain-prebuilt.git
and get file simulator-example-Sha3RocketConfig-debug
but when I ran sha3-rocc.c example using:
./simulator-example-Sha3RocketConfig-debug +verbose -v sha3.vcd /mnt/c/chipyard/generators/sha3/src/main/c/sha3.rv
I got errors (hwacha has the same problem):
./simulator-example-Sha3RocketConfig-debug +verbose -v sha3.vcd /mnt/c/chipyard/generators/sha3/src/main/c/sha3.rv
using random seed 1578896255
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 63817
C0:          0 [0] pc=[fffffff00781e10f] W[r 0=f4127e83c66bb0c8][0] R[r24=0000000000000000] R[r15=5248c5aa209602f5] inst=[00007462] DASM(00007462)
warning: tohost and fromhost symbols not in ELF; can't communicate with target
thanks!

colinschmidt, 2020-01-14:
How did you compile the c code? I don't think that test is a baremetal test so you would need to use something like `pk` as mentioned in the sha3 readme.
Based on your path it seems like you are using an old binary that just happens to be committed.

You should try using the steps in the readme here:
https://github.com/ucb-bar/sha3-workload/tree/63e154a6e00cc6c977eaac8fbaa59c30ceb67562

dreamflyings, 2020-01-15:
The toolchain in the chipyard project is particularly slow to download in China, so I used a toolchain from another project eg. "hwacha-template"(riscv-tools is a version of esp-tools,I changed a few esp-tools versions, but all have such problems).So I need to download the original toolchain to prove the sh3 is right, but this error on hwacha has been difficult to solve.

dreamflyings, 2020-01-15:
Can be modified as follows in "variables.mk"
ifeq ($(SUB_PROJECT),hwacha)
	SBT_PROJECT       ?= example
	MODEL             ?= TestHarnessWithDTM
	VLOG_MODEL        ?= TestHarnessWithDTM
	MODEL_PACKAGE     ?= example
	CONFIG            ?= HwachaRocketConfig
	CONFIG_PACKAGE    ?= example
	GENERATOR_PACKAGE ?= hwacha
	TB                ?= TestDriver
	TOP               ?= TopWithDTM
endif
make debug SUB_PROJECT=hwacha
After setting it up like that the previous BUG disappeared.
(./simulator-example-HwachaRocketConfig-debug +verbose -v hello.vcd /mnt/c/hello.riscv :: this is right)
but another problem has come, when I ran a hwacha program. The error is shown below（Sha3's the same）.
<--::
C0:       4320 [0] pc=[0000000000000814] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r20=0000000000000003] inst=[f1402473] DASM(f1402473)
e, but allowing simulation to continue and fail.ERROR: ../fesvr/dtm.cc:360, Debug Abstract Command Error #3 (EXCEPTION)ERROR: ../fesvr/dtm.cc:361, Should die, but allowing simulation to continue and fail.ERROR: ../fesvr/dtm.cc:360, Debug Abstract Command Error #3 (EXCEPTION)ERROR: ../fesvr/dtm.cc:361, Should die, but allowing simulation to continue and fail.ERROR: ../fesvr/dtm.cc:360, Debug Abstract Command Error #3 (EXCEPTION)ERROR: ../fesvr/dtm.cc:361, Should die, but allowing simulation to continue and fC0:       4321 [0] pc=[0000000000000814] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r20=0000000000000003] inst=[f1402473] DASM(f1402473)
::-->

colinschmidt, 2020-01-15:
Can you please try using these steps for sha3?
https://github.com/ucb-bar/sha3-workload/blob/master/README.md

dreamflyings, 2020-01-16:
@colinschmidt  Thank you for your reply.The purpose of testing this is mainly to make the hwacha coprocessor run on FPGA.So I hope to see the waveform for further research.SH3 is used for Comparative Test.The tools you provided are a bit complicated  and I'm not sure if it can meet my needs. After the Chinese New Year  I will spend some time learning this tool.

colinschmidt, 2020-01-16:
You will need to use the esp-tools built from chipyards build-toolchain script to run hwacha code.
You should also use the makefile to run the test to make sure you don't miss any arguments that need to be passed to the simulator.

If you follow those steps and it doesn't work feel free to post here again with the errors you get.

alonamid, 2020-04-29:
Closed due to inactivity. Feel free to re-open if necessary.

closed 2020-04-29 by alonamid