ucb-bar/chipyard #377: Most recently changed print statements aren't included when compiling

baltazarortiz, 2020-01-14:
This seems to be the same issue as #319, but as there was no information provided about replicating and no details as to why it was closed shortly after opening, I figured a new issue would be better.

I also tried creating a new generator and noticed that printf outputs weren't showing up when I expected, but as I added more printf calls, earlier calls that hadn't shown up before began to show up. I somewhat replicated this on master using the pwm example generator, though it seems like something is kind of nondeterministic.

I first changed the `PWMBase` class to print the PWM output:

```
  // If PWM is enabled, pwmout is high when counter < duty                      
  // If PWM is not enabled, it will always be low                               
  val pwmval = Wire(Bool())                                                     
  pwmval := io.enable && (counter < io.duty)                                    
  printf(p"pwmval: $pwmval \n")                                        
```

After running a clean build (`make clean; make CONFIG=PWMRocketConfig TOP=TopWithPWMTL -j32`), running the test with `./simulator-example-PWMRocketConfig +verbose ../../tests/pwm.riscv` did not show the expected printf output (including or not including `+verbose` did not make a difference either way). However, when I went through my steps one more time to write up this issue, it did show up, and I don't believe there was any difference.

Adding another printf beneath the first call (I just used `printf(p"another print\n")`) and recompiling resulted in the `pwmval:` output displaying, but the `another print` line not being shown.

Similarly, adding a third printf beneath the second line and recompiling resulted in the first two printfs appearing in the output. (In the redo, it did not show up, but adding a fourth printf caused the second and third message to output).

Commenting out or removing lines suffers from the same issue - the next compilation still includes the printed output from the line that should no longer have any effect. 

Is this an issue with the build system for chipyard? When I add and remove print statements in a non-chipyard Chisel project, the change is immediately reflected when running tests.

abejgonzalez, 2020-01-16:
Hmm. I went ahead and followed your instructions on `dev` and I was able to see all printfs at all times. Are you able to see the `printfs` in the verilog output (`fwrites`) when you update the code and recompile?

baltazarortiz, 2020-01-16:
I tried adding a fourth print and it immediately showed up in the verilog and simulator output, but when I added a fifth (below all the others, not in a different module/logic path or anything), it didn't show up in the verilog/output. I did see it in the FIR file though:

```
example.TestHarness.PWMRocketConfig.fir                                         
222333:      printf(clock, UInt<1>(1), "fifth print\n") @[PWM.scala 41:9] 
```

baltazarortiz, 2020-01-16:
I also tried compiling without the `-j32` flag in case there was some multicore compilation related bug, but no luck and the next two prints I've added haven't shown up, even in FIR. Trying a single core compilation after a `make clean` just in case that helps. I'll try to think of other things to test in this repo, and will try to replicate on a freshly pulled repo with separately built tools!

abejgonzalez, 2020-01-16:
> I tried adding a fourth print and it immediately showed up in the verilog and simulator output, but when I added a fifth (below all the others, not in a different module/logic path or anything), it [...]

Interesting. If it shows up in that `example.TestHarness.PWMRocketConfig.fir` file then it should show up in the `*.top.fir` file (located in the same place `generated-src`). If that is not the case, then maybe the `barstools.tapeout.transforms.GenerateTopAndHarness` run is not getting redone (which might be a make issue)?

A couple things:
1. Can you checkout the `dev` branch of the repo and try doing the same tests (adding prints)
2. Can you post a build log where you just change one line, save, run the `make` command, and the prints don't show up?

baltazarortiz, 2020-01-16:
Sure thing! Also re: multicore checking I think I may have targeted the wrong folder so they may have actually showed up, so I'll check back on that.

baltazarortiz, 2020-01-16:
In the process of a clean clone to try `dev`, but in the meantime, here's a build log. I put in the first print and it showed up on the first compile, but the second print only showed up in the FIR file again. 

search results:
```
$ ag "second print"
example.TestHarness.PWMRocketConfig.fir
222318:      printf(clock, UInt<1>(1), "second print\n") @[PWM.scala 38:9]
```

build log (`make CONFIG=PWMRocketConfig TOP=TopWithPWMTL -j32`):
```
mkdir -p /home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig
cd /home/ubuntu/chipyard && java -Xmx8G -Xss8M -XX:MaxPermSize=256M -jar /home/ubuntu/chipyard/generators/rocket-chip/sbt-launch.jar ++2.12.4 "project example" "runMain example.Generator /home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig example TestHarness example PWMRocketConfig"
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[0m[[0m[0minfo[0m] [0m[0mLoading global plugins from /home/ubuntu/.sbt/1.0/plugins[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project chipyard-build from plugins.sbt,metals.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading project definition from /home/ubuntu/chipyard/project[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project dsptools from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project barstoolsMacros from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project mdf from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project boom from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project hwacha from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project icenet from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project testchipip from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project hardfloat from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project chisel_testers from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project treadle from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project firrtl_interpreter from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project chisel from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project chipyard from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project sim-build from plugins.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading project definition from /home/ubuntu/chipyard/sims/firesim/sim/project[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project firesim from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project midas from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mLoading settings for project targetutils from build.sbt ...[0m
[0m[[0m[0minfo[0m] [0m[0mResolving key references (23770 settings) ...[0m
[0m[[0m[0minfo[0m] [0m[0mSet current project to chipyard (in build file:/home/ubuntu/chipyard/)[0m
[0m[[0m[0minfo[0m] [0m[0mSetting Scala version to 2.12.4 on 30 projects.[0m
[0m[[0m[0minfo[0m] [0m[0mReapplying settings...[0m
[0m[[0m[0minfo[0m] [0m[0mSet current project to chipyard (in build file:/home/ubuntu/chipyard/)[0m
[0m[[0m[0minfo[0m] [0m[0mSet current project to example (in build file:/home/ubuntu/chipyard/)[0m
[0m[[0m[33mwarn[0m] [0m[0mMultiple main classes detected.  Run 'show discoveredMainClasses' to see the list[0m
[0m[[0m[33mwarn[0m] [0m[0mMultiple main classes detected.  Run 'show discoveredMainClasses' to see the list[0m
[0m[[0m[33mwarn[0m] [0m[0mMultiple main classes detected.  Run 'show discoveredMainClasses' to see the list[0m
[0m[[0m[0minfo[0m] [0m[0mCompiling 1 Scala source to /home/ubuntu/chipyard/generators/example/target/scala-2.12/classes ...[0m
[0m[[0m[33mwarn[0m] [0m[0mthere were 10 feature warnings; re-run with -feature for details[0m
[0m[[0m[33mwarn[0m] [0m[0mone warning found[0m
[0m[[0m[0minfo[0m] [0m[0mDone compiling.[0m
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.protobuf.UnsafeUtil (file:/home/ubuntu/.sbt/boot/scala-2.12.7/org.scala-sbt/sbt/1.2.8/protobuf-java-3.3.1.jar) to field java.nio.Buffer.address
WARNING: Please consider reporting this to the maintainers of com.google.protobuf.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[0m[[0m[0minfo[0m] [0m[0mPackaging /home/ubuntu/chipyard/generators/example/target/scala-2.12/example_2.12-1.0.jar ...[0m
[0m[[0m[0minfo[0m] [0m[0mDone packaging.[0m
[0m[[0m[0minfo[0m] [0m[0mRunning example.Generator /home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig example TestHarness example PWMRocketConfig[0m
[[35minfo[0m] [0.009] Elaborating design...
Interrupt map (2 harts 2 interrupts):
  [1, 2] => TestHarness

L2 InclusiveCache Client Map:
	0 <= Core 0 DCache
	1 <= Core 0 ICache
	2 <= slave-port-axi4 ID#0
	3 <= slave-port-axi4 ID#1
	4 <= serial

<stdout>: Warning (simple_bus_reg): Node /soc/external-interrupts missing or empty reg/ranges property
/dts-v1/;

[... 112 lines cut ...]
	   10000 -    20000  R X  rom@10000
[... cut ...]
[[33mwarn[0m] Line numbers for deprecations reported by Chisel may be inaccurate; enable scalac compiler deprecation warnings via either of the following methods:
[[33mwarn[0m]   In the sbt interactive console, enter:
[[33mwarn[0m]     set scalacOptions in ThisBuild ++= Seq("-unchecked", "-deprecation")
[[33mwarn[0m]   or, in your build.sbt, add the line:
[[33mwarn[0m]     scalacOptions := Seq("-unchecked", "-deprecation")
[[35minfo[0m] [22.003] Done elaborating.
[0m[[0m[32msuccess[0m] [0m[0mTotal time: 40 s, completed Jan 16, 2020, 2:52:15 PM[0m
```

baltazarortiz, 2020-01-16:
Having some issues with getting `dev` to compile that are unrelated to this, but I'll try to update on that soon!

abejgonzalez, 2020-01-16:
Hmm. The `makefile` should notice that the main `<long-name>.fir` file changed. Here are the relevant makefile locations:

https://github.com/ucb-bar/chipyard/blob/61ca6fa294526bb20443f15766a374c01071b0b8/common.mk#L57-L60
^ That shows that `firrtl_temp` depends on the `FIRRTL_FILE` aka the `<long-name>.fir`. So if that file changes then it should rerun the `GenerateTopAndHarness` step.

https://github.com/ucb-bar/chipyard/blob/61ca6fa294526bb20443f15766a374c01071b0b8/common.mk#L42-L45
^ That shows how the `FIRRTL_FILE` is generated.

Maybe add some prints in those areas to double-check that dependencies are getting tracked properly or use `make -d` to see all dependencies?

For reference, I am using `make` 4.1.

abejgonzalez, 2020-01-16:
Also, I remember that making the `makefile` `-j` compatible turned out to be a bit tricky. So that definitely could be the issue. Though, I'm surprised my +5-10 tests haven't failed. Maybe I am getting lucky?

baltazarortiz, 2020-01-20:
Thanks! I'll dig into this a bit more and see if I can figure anything out.

baltazarortiz, 2020-01-20:
Messed around with things a bit more, and it's definitely a multicore issue - it works every time with single core compilation. If `-j` support is not a priority and this is not an obvious fix I can stick with single core compilation, though I'd love to get tips on speeding up builds.

I ran `make clean`, then added a single print statement and reran a build using `-j32`. This worked fine - the print showed up in the final verilog.

I believe the relevant snippets of the log are:

```
Must remake target '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/sim_files.f'.    
Need a job token; we don't have children       
cd /home/ubuntu/chipyard && java -Xmx8G -Xss8M -XX:MaxPermSize=256M -jar /home/ubuntu/chipyard/generators/rocket-chip/sbt-launch.jar ++2.12.4 "project utilities" "runMain utilities.GenerateSimFiles -td /home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig -sim verilator" 
Putting child 0x559cd19871e0 (/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/sim_files.f) PID 9692 on the chain
            Recipe of '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/sim_files.f' is being run. 

...

     File '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/example.TestHarness.PWMRocketConfig.anno.json' does not exist.

...

[this part shows up six times?]
Considering target file 'firrtl_temp'.
 File 'firrtl_temp' does not exist.
  Pruning file '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/example.TestHarness.PWMRocketConfig.fir'. 
  Pruning file '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/example.TestHarness.PWMRocketConfig.anno.json'.  
 Finished prerequisites of target file 'firrtl_temp'.                                                                                           
The prerequisites of 'firrtl_temp' are being made.    
```

Then, I added a second print statement and ran a second multicore build, but the print only showed up in the FIR file, not the verilog. A diff showed that what I believe were the equivalent lines to the first log were slightly different. There are a lot of messages like this:

```
Prerequisite '/home/ubuntu/chipyard/generators/utilities/src/main/scala/Simulator.scala' is older than target '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/sim_files.f'.                                                                                                                                                                                                                                                       
Prerequisite '/home/ubuntu/chipyard/generators/utilities/src/main/scala/TestSuites.scala' is older than target '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/sim_files.f'.                                                                                                                                                                                                                                                      
Prerequisite '/home/ubuntu/chipyard/generators/utilities/src/main/scala/System.scala' is older than target '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/sim_files.f'.                                                                                                                                                                                                                                                          
Prerequisite '/home/ubuntu/chipyard/generators/utilities/src/main/scala/Subsystem.scala' is older than target '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/sim_files.f'.                                                                                                                                                                                                                                                       
Prerequisite '/home/ubuntu/chipyard/lib/firrtl.jar' is older than target '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/sim_files.f'.     
 No need to remake target '/home/ubuntu/chipyard/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/sim_files.f'.                     
```
Of potential interest, there is a message saying:
```
No recipe for '/home/ubuntu/chipyard-ref-pipe/sims/verilator/generated-src/example.TestHarness.PWMRocketConfig/example.TestHarness.PWMRocketConfig.fir' and no prerequisites actually changed.
```

Could this be the issue?

In case I missed something, I'll find a spot to upload the full logs (~60MB each) so you can check out a diff.

I still haven't been able to get `dev` working, but I'm not sure if debugging that should go here or in a separate issue.

abejgonzalez, 2020-01-21:
Not getting `dev` working should be a separate issue but probably not super important since we are going to move `dev` to `master` _hopefully_ this week.

From what I noticed, multi-core doesn't help RTL builds much. The only thing that make parallelizes is the `MacroCompiler` run. The rest is fairly serial.

I'll try to look into this soon. Thanks for narrowing down the issue.

abejgonzalez, 2020-01-21:
Ok for reference, here is a smaller reproducible snippet:

```
.INTERMEDIATE: temp
default: f

a b c:  temp

temp: i1 i2
    echo "BUILD: a b c"
    cat i1 i2 > a
    cat i1 i2 > b
    cat i1 i2 > c

e: a b c
    echo "BUILD: e"
    touch $@

f: e
    echo "BUILD: f"
    touch $@
```

And the reference to the main Stack Overflow post I was working off of when making the initial changes (https://stackoverflow.com/questions/19822435/multiple-targets-from-one-recipe-and-parallel-execution/).

I think the solution is to add `touch $@` under the `a b c: temp` like the following:

```
a b c: temp
    touch $@
```

Im doing more tests as we speak, but go ahead and try out #386 (note: this is referencing the `dev` branch).

abejgonzalez, 2020-01-22:
Solved by #386

closed 2020-01-22 by abejgonzalez

baltazarortiz, 2020-01-22:
Thanks for checking this out! I'll see if I can get `dev` working, otherwise I'll wait till it gets merged in.

PR #386 (2020-01-22): Patch parallel make
Occasionally, parallel `make` will silently ignore source file changes. See #377 and https://stackoverflow.com/questions/19822435/multiple-targets-from-one-recipe-and-parallel-execution/. This attempts to address this issue by adding a `touch` to the targets depending on the `INTERMEDIATE` value.

This was tested by doing `make -j32` on the config./system listed in #377. 

Fixes #377