ucb-bar/chipyard #420: Currently version of QEMU needs Capstone header fix.

jsaunders-TDKC, 2020-01-31:
Attempting to compile the RISC-V tools with the "./scripts/build-toolchains.sh" script fails when the file "toolchains/qemu/include/disas/capstone.h" can't find the include file "<capstone.h>". This needs to be changed to "<capstone/capstone.h>" in order for compilation to succeed as determined by reading:
https://github.com/aquynh/capstone/issues/1339
https://bugs.archlinux.org/task/61507

This doesn't appear to be in upstream QEMU in the master branch either, so it may be more appropriate to open a bug there.

NathanTP, 2020-05-01:
Which OS is this on? It still works on the Centos-based AWS FPGA images. We primarily use centos 7 to develop.

A workaround might be to try and downgrade capstone or figure out which versions are compatible.

jsaunders-TDKC, 2020-05-04:
This was on Arch Linux back when I first reported it (capstone 4.0.1+, going by the Arch Linux bug report). At the moment, I'm preferring the workaround of editing the one line of code as mentioned above to get it working, so I don't have to mess around with downgrading capstone and breaking other packages or having to use a copy of capstone specific to this.

NathanTP, 2020-05-04:
This workaround seems reasonable. The issues seems to be with either the distros or upstream Qemu. Since things work on our default distros (centos 7 and Ubuntu) I don't think we're in a position to fix it.

closed 2020-05-04 by NathanTP