ucb-bar/chipyard #422: add a spi flash config

siyy123, 2020-02-03:
**Impact**: rtl

**Description**

Can we add a spi flash example? since in real tapeout the chip may need a external storage to store firmware, kernel, os and software.

jwright6323, 2020-04-29:
This would be a good thing to add, but we don't currently have access to any open-source verification IP for SPI flash.

colinschmidt, 2020-05-04:
I think we could add a config fragment that doesn't have any test harness support.
@siyy123 Would that be helpful?

jwright6323, 2020-05-04:
I haven't really looked at this much, but there appears to be a freely available QSPI model here: http://ww1.microchip.com/downloads/en/DeviceDoc/sst26wf016b.v

jwright6323, 2020-05-04:
Also: https://www.micron.com/products/nor-flash/serial-nor-flash/part-catalog/n25q00aa13g1240e

We can look into licenses to see if either are compatible with ours.

jwright6323, 2020-05-04:
I looked into these two things a bit, and unfortunately they won't work with Verilator because they have # delays. We'll probably have to build our own model.

a0u, 2020-05-04:
There are Verilog models from ISSI: http://www.issi.com/US/product-flash.shtml
The HiFive-1 board uses the IS25LP128-JBLE part.

N25Q from Micron is being replaced with the MT25Q family, but on a cursory glance, the newer models don't appear to support zero-delay simulation either.

jwright6323, 2020-05-15:
This now added into dev with #546 . It will be in the Chipyard 1.3 release.

closed 2020-05-15 by jwright6323