ucb-bar/chipyard #424: Stage/Phase Migration

seldridge, 2020-02-04:
**Impact**: other

**Description**

@debs-sifive has been working on moving Rocket Chip to Stage/Phase infrastructure (see: https://github.com/chipsalliance/rocket-chip/pull/2274). This likely has implications for Chipyard or any projects which define new generators. As Chipyard is a big stakeholder, it would be good to have inputs from Chipyard on how this should work and/or a joint jump to Stage/Phase.

This feature request is both:

1. An inquiry into how complicated it would be to migrate Chipyard to Stage/Phase
2. A request for this to be done

This is then intended to provide input to Rocket Chip Stage/Phase PRs related to:

1. Is it safe to change the command line API of Rocket Chip?
1. Can the existing `Generator` be removed?
1. What should an expected deprecation schedule look like for removing the `Generator`?
1. How should additional test suites be specified to the Rocket Chip stage?

**What is a motivating example for changing the behavior?**

1. Migrate to new infrastructure
1. The usual stage/phase pitch of switching the generation process to composition of functions 
1. The ability to use the dependency API

colinschmidt, 2020-02-04:
I've already done this in the past to use aspects. It was a pain to hack on rocket-chip so it was shelved until that was fixed. Now that it has been (thanks you two) fixed I expect to get back to it.

Specific responses:
1. I actually wanted to change the command line API and was annoyed at rocket-chip's rigid requires limiting the extensibility of the command line args.
2. I believe the answer is yes, and for stage phase it was actually getting in the way. We have several downstream generators but all of them would probably be able to handle the change (some may appreciate a less rigid structure).
3. This question is harder but since rocket-chip doesn't really have releases it doesn't seem super relevant. Chipyard has releases every few months but has not made backwards compatibility guarantees so it may not truly need a schedule but we will discuss that at our next meeting.
4. This is a good question that I don't have a good answer to yet.

alonamid, 2020-04-29:
I believe most have these were addressed by @davidbiancolin and resolved by #503

closed 2020-04-29 by alonamid