ucb-bar/chipyard #430: Issue with building a DTM+JTAG

rajeshkrmeena, 2020-02-13:
I build with "make CONFIG=jtagRocketConfig". It generated the DTM-JTAG module with ROCKET. But, when I tried to simulate it but it returning busy when trying to read/write a DM register except for DMCONTROL (0x10) register. I looked into inside also found that dmi_req_resp_valid is not return /asserted if accessing a DM register except DMCONTROL. 

A screenshot, debug module,  is attached.
[image]

colinschmidt, 2020-02-20:
Are you connecting to it via openocd? 
Did you follow the steps here: 
https://chipyard.readthedocs.io/en/latest/Advanced-Concepts/Chip-Communication.html#using-the-jtag-interface

rajeshkrmeena, 2020-02-24:
Yes, I want to connect it via Openocd. Yes, I followed all the steps.

closed 2020-12-07 by alonamid