ucb-bar/chipyard #431: configure.ac is required

OmarDez, 2020-02-13:
I run the command `./scripts/build-toolchains.sh riscv-tools` and the next thing is 
`autoreconf: 'configure.ac' or 'configure.in' is required`
The env.sh is nor emitted what to do?

OmarDez, 2020-02-13:
when running `./scripts/init-submodules-no-riscv-tools.sh` crashes in `cloning into 'env'`, `connection reser by peer`

abejgonzalez, 2020-02-14:
This seems like you repository is somehow messed up. What version of Chipyard do you have (hash) / modifications?

OmarDez, 2020-02-14:
the default when I run the commands from the docs page, I install every module of the toolchain and it worked, but when building the example in verilator it show some errors
The broken git link was in torture example of rocket chip

alonamid, 2020-04-29:
This seems to be a git issue. 
Closing due to inactivity. Feel free to re-open if there is additional follow up with additional information.

closed 2020-04-29 by alonamid