ucb-bar/chipyard #436: RISCV Toolchain Build Error

rayjyh, 2020-02-17:
Hi, I got the following error when building the riscv toolchain by running ./build-toolchains.sh. 
[log.txt](https://github.com/ucb-bar/chipyard/files/4214349/log.txt)

jingyuhe@jingyuhe-VirtualBox:~/Desktop/chipyard/scripts$ ./build-toolchains.sh > log.txt
+ set -e
+ set -o pipefail
++ pwd
+ RDIR=/home/jingyuhe/Desktop/chipyard/scripts
++ git rev-parse --show-toplevel
+ CHIPYARD_DIR=/home/jingyuhe/Desktop/chipyard
+ TOOLCHAIN=riscv-tools
+ EC2FASTINSTALL=false
+ RISCV=
+ '[' '' '!=' '' ']'
+ '[' -z '' ']'
+ INSTALL_DIR=riscv-tools-install
++ pwd
+ RISCV=/home/jingyuhe/Desktop/chipyard/scripts/riscv-tools-install
+ echo 'Installing toolchain to /home/jingyuhe/Desktop/chipyard/scripts/riscv-tools-install'
+ export RISCV=/home/jingyuhe/Desktop/chipyard/scripts/riscv-tools-install
+ RISCV=/home/jingyuhe/Desktop/chipyard/scripts/riscv-tools-install
+ cd /home/jingyuhe/Desktop/chipyard
++ pwd
+ SRCDIR=/home/jingyuhe/Desktop/chipyard/toolchains/riscv-tools
+ '[' -d /home/jingyuhe/Desktop/chipyard/toolchains/riscv-tools ']'
+ . ./scripts/build-util.sh
++ '[' -n /home/jingyuhe/Desktop/chipyard/toolchains/riscv-tools ']'
+++ getconf _NPROCESSORS_ONLN
++ ncpu=1
++ case ${ncpu} in
++ export 'MAKEFLAGS=-j 1 -j8'
++ MAKEFLAGS='-j 1 -j8'
+++ command -v make
++ MAKE=/usr/local/bin/make
++ readonly MAKE
+ '[' false = true ']'
+ module_prepare riscv-gnu-toolchain qemu
+ set -e
+ name=riscv-gnu-toolchain
+ shift
+ dir=/home/jingyuhe/Desktop/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain
+ echo '=>  Starting riscv-gnu-toolchain build'
+ echo '==>  Initializing riscv-gnu-toolchain submodule'
+ '[' 1 -gt 0 ']'
+ git submodule update --init /home/jingyuhe/Desktop/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain
+ '[' -n qemu ']'
+ git -C /home/jingyuhe/Desktop/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain config submodule.qemu.update none
+ shift
+ '[' -n '' ']'
+ git submodule update --init --recursive /home/jingyuhe/Desktop/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain
+ module_build riscv-gnu-toolchain --prefix=/home/jingyuhe/Desktop/chipyard/scripts/riscv-tools-install --with-cmodel=medany
+ set -e -o pipefail
+ name=riscv-gnu-toolchain
+ shift
+ cd /home/jingyuhe/Desktop/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain
+ '[' -e build ']'
+ echo '==>  Removing existing riscv-gnu-toolchain/build directory'
+ rm -rf build
+ '[' -e configure ']'
+ mkdir -p build
+ cd build
+ tee build.log
jingyuhe@jingyuhe-VirtualBox:~/Desktop/chipyard/scripts$ vim log.txt 

My environment is 
jingyuhe@jingyuhe-VirtualBox:~/Desktop/chipyard/scripts$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.6 LTS
Release:	14.04
Codename:	trusty
jingyuhe@jingyuhe-VirtualBox:~/Desktop/chipyard/scripts$

kvtam, 2020-03-04:
I'm encountering the same error.

jerryz123, 2020-03-04:
From the error log, it looks like you did not follow all steps in https://chipyard.readthedocs.io/en/dev/Chipyard-Basics/Initial-Repo-Setup.html

Note the required dependencies.

closed 2020-04-29 by alonamid