ucb-bar/chipyard #454: error: undefined reference to pthread_once in toolchains building

AlethTan, 2020-02-27:
I cloned from github: 

- riscv-boom
- chipyard

./scripts/build-toolchains.sh riscv-tools or ./scripts/build-toolchains.sh riscv-tools esp-tools sent back the same error: 
`
builddir/build/BUILD/gcc-build/x86_64-generic-linux/libstdc++-v3/include/x86_64-generic-linux/bits/gthr-default.h:700 : undefined reference to « pthread_once »
`
`collect2: error: ld returned 1 exit status
gmake[2]: *** [Makefile:2155 : lto-wrapper] Erreur 1
`
`
[Makefile:4324 : all-gcc] Erreur 2
`
`
[Makefile:432 : stamps/build-gcc-newlib-stage1] Erreur 2
`
After a deep search, it seems to be a cross-compiled problem or a gcc version.
I installed c-basic-legacy (32 bits) and c-basic-static. 
But nothing helps.
OS: clear linux - Intel core Xeon 12 cores -  gcc 9
May I benefit from your help?

a0u, 2020-02-27:
This seems to be a header mismatch problem with native gcc.  Do you have `C_INCLUDE_PATH` set in your environment that would cause gcc to search a non-standard path before the system search paths?

`builddir/build/BUILD/gcc-build/x86_64-generic-linux/libstdc++-v3/include/x86_64-generic-linux/bits/gthr-default.h` appears to be a header from a source or staging directory that might not match the gcc being used.  Normally I would expect the system-wide header to be installed in `/usr/include/x86_64-linux-gnu/c++/5/bits/gthr-default.h` or similar (but I am not familiar with the Clear Linux distro).

If possible, could you attach:
* `chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build.log`
* `chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-gcc-newlib-stage1/config.log`

AlethTan, 2020-02-27:
See enclosed the 2 files you asked for.
It's weird, some checking sent back gnat and its dependencies are missing, but gnat is installed. Same for valgrind.
I found gthr-default.h under `/usr/include/c++/9/x86_64-generic-linux/bits`
I put the path in `.bashrc`, still the same `undefined reference to « pthread_once »`
[build.log](https://github.com/ucb-bar/chipyard/files/4264380/build.log)
[config.log](https://github.com/ucb-bar/chipyard/files/4264386/config.log)

a0u, 2020-02-27:
A curious discrepancy is that the system libraries (libstdc++, libgcc) appear to link against pthread functions, judging by the link error, but your compiler seems to have been built without threading support.  One of the configure tests (for libbacktrace in gcc) shows:
```
checking whether -pthread is supported... no
```

Are you able to link against libpthread?
```sh
echo 'int main(void) { return 0; }' | g++ -xc++ -lpthread -o /dev/null -
```

By the way, you don't need to add `/usr/include/c++/9/x86_64-generic-linux/bits` to your PATH.  I was referring to a different environment variable (`C_INCLUDE_PATH`).

AlethTan, 2020-02-28:
I installed `devpkg-libpthread-stubs`
In the build.log file:

- checking for pthread.h... yes
- checking whether -pthread is supported... yes

Still the same error.
Yes, the link against lipthread exists.

AlethTan, 2020-02-28:
I set the environment variable: 
`C_INCLUDE_PATH:/usr/include/c++/9/x86_64-generic-linux/bits/`
`export C_INCLUDE_PATH`
`echo $C_INCLUDE_PATH`
`/usr/include/c++/9/x86_64-generic-linux/bits/`
The same error backs.
Why the environment variable is not taking into account during compilation?
The build.log is enclosed.
May I ask you a little more of your time to help me move forward?
[build.log](https://github.com/ucb-bar/chipyard/files/4266656/build.log)