ucb-bar/chipyard #455: recipe for target 'firrtl_temp' failed

umair-riaz, 2020-02-28:
Hi,
I have gone through the setup and installation of Chipyard and after installation, it passed all make run-asm-tests and make run-bmark-tests but got an error which running the following command.

Command:
>> make CONFIG=SmallBoomAndRocketConfig

Error:
/chipyard/common.mk:68: recipe for target 'firrtl_temp' failed

If someone have gone through it, kindly reply.

abejgonzalez, 2020-03-02:
Can you please post a build log starting from the command run to the error, please? Normally, when this breaks, the FIRRTL compile didn't pass. Did you make any changes to the code? Version of Chipyard?

umair-riaz, 2020-03-05:
I didn't make any changes but got error. I have attached the build log starting from the command run.

[build_log.txt](https://github.com/ucb-bar/chipyard/files/4292114/build_log.txt)

KlepD-SAL, 2020-03-09:
Hi,

any updates on the topic?
I also have the same problem when running the default example with `make`.

Following error occurs:
```
java -Xmx8G -Xss8M -XX:MaxPermSize=256M -jar ~/Documents/chipyard/generators/rocket-chip/sbt-launch.jar ++2.12.10 "project tapeout" "runMain barstools.tapeout.transforms.GenerateTopAndHarness -o ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.top.v -tho ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.harness.v -i ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.fir --syn-top Top --harness-top TestHarness -faf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.anno.json -tsaof ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.top.anno.json -tdf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/firrtl_black_box_resource_files.top.f -tsf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.top.fir -thaof ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.harness.anno.json -hdf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/firrtl_black_box_resource_files.harness.f -thf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.harness.fir --infer-rw --repl-seq-mem -c:TestHarness:-o:~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.top.mems.conf -thconf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.harness.mems.conf -td ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig"
~/Documents/chipyard/common.mk:68: recipe for target 'firrtl_temp' failed
make: *** [firrtl_temp] Error 137
```

I tried it with the most up-to-date version from the repo as well as release version 1.1.0.
Makes no difference.

Any idea?

OS: Ubuntu 18.04

Cheers

KlepD-SAL, 2020-03-11:
@umair-riaz 

I managed to resolve my problems.
My guess is, that there was a problem with the dependencies.
Closely following the [Initial Repository Setup](https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html) made it work, although I think that there is a typo in the Ubuntu dependencies.
**libxpat1-dev** was not available in the repo, so I installed **libexpat1-dev**.
Maybe `scripts/ubuntu-req.sh` should be updated if this really is a typo.

CoCoMilkyWay, 2020-04-13:
> I didn't make any changes but got error. I have attached the build log starting from the command run.
> 
> [build_log.txt](https://github.com/ucb-bar/chipyard/files/4292114/build_log.txt)

maybe a pic is more readable :)
I guess I have similar problem like you
[image]

CoCoMilkyWay, 2020-04-15:
@KlepD-SAL @umair-riaz 
I think I figured this out, this is because my computer sucks (not enough memory)
[image]
Let’s say for SmallBoomConfig, you need at least 7Gb ram in Ubuntu or there will be firrtl_temp error
For LargeBoomAndRocketConfig, you need 11Gb ram in Ubuntu

@abejgonzalez
Also can you please verify this and maybe put this in the Chipyard document? This is very confusing to ppl learning how to use Chipyard and could eat out a lot of their time trying to debug :)

abejgonzalez, 2020-04-15:
Good find. Yes, BOOM configs normally do take up a significant amount of RAM (ex. `DualLargeBoomAndDualRocketConfig` took about 9GBs for me right now). Normally, this is not an issue on the UCB machines but could definitely be an issue on other computers.

I'm not exactly sure if the documentation is the best place to put this. Most of the documentation is set up to be like a tutorial or a description of what Chipyard does/contains. I think having just the GH issue is fine (it looks like most Chipyard issues are getting resolved/answered as GH issues since we can interact with you here).

KlepD-SAL, 2020-04-15:
@CoCoMilkyWay 
I ran into the same problem and wanted to tell you that the memory is maybe the problem.
But you were faster ;)

@abejgonzalez 
Is it possible to print out a (more) clear error message when running into the error with low memory?
If not, I also suggest to mention it in the documentation somewhere. Maybe "1.4. Initial Repository Setup -> 1.4.1. Requirements" would be a good place to start since basically everyone setting up an environment will start there.

abejgonzalez, 2020-04-16:
> Is it possible to print out a (more) clear error message when running into the error with low memory?

I don't think so. While `Killed` isn't very descriptive I think it leads to more general system-level problems that each user has to individually look up (since it could be a multitude of things). 

> If not, I also suggest to mention it in the documentation somewhere. Maybe "1.4. Initial Repository Setup -> 1.4.1. Requirements" would be a good place to start since basically everyone setting up an environment will start there.

Sure. Feel free to make a PR to the `dev` branch on Chipyard of where you think it would make sense to go and Ill be sure to review it!

KlepD-SAL, 2020-04-16:
Sorry for the question, but I am new to open-source contributions:
Should I fork the project and modify the doc there or can I just create a branch on this repository?

abejgonzalez, 2020-04-16:
Yes. Feel free to fork the project and modify the doc there. Then, push the branch to GH (in your fork). Afterwards, you can create a new PR with the changes.

KlepD-SAL, 2020-04-17:
Hi, I added some changes and made the pull request: #517

closed 2020-05-04 by abejgonzalez