ucb-bar/chipyard #458: Error simulating default examples with verilator

rayjyh, 2020-03-04:
I encountered the following error when running `make` in /sims/verilator.

`chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/SimDTM.cc:3:10: fatal error: fesvr/dtm.h: No such file or directory
 #include <fesvr/dtm.h>
          ^~~~~~~~~~~~~
compilation terminated.
VTestHarness.mk:69: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory '/home/jyh/Desktop/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig'
Makefile:101: recipe for target '/home/jyh/Desktop/chipyard/sims/verilator/simulator-example-RocketConfig' failed
make: *** [/home/jyh/Desktop/chipyard/sims/verilator/simulator-example-RocketConfig] Error 2`

fesvr/dtm.h is at `/chipyard/toolchains/riscv-tools/riscv-isa-sim/fesvr/`. According to https://github.com/chipsalliance/rocket-chip, we are supposed to `export RISCV=/path/to/riscv/toolchain/installation` before building the toolchain, which I suppose is /chipyard/toolchains/riscv-tools . But `/scripts/build-toolchains.sh` set `$RISCV` to be `/chipyard/riscv-tools-install`. I tried to manually modify `$RISCV` as discussed in https://github.com/chipsalliance/rocket-chip/issues/1448, the error still persists.

colinschmidt, 2020-03-04:
Did your toolchain build fail?
`$RISCV` should be pointing to `chipyard/riscv-tools-install`. 
Is there a `fesvr/dtm.h` in `chipyard/riscv-tools-install/include`?

rayjyh, 2020-03-04:
The build was successful. And there is `fesvr/dtm.h` in `riscv-tools-install/include`. But it still failed.

colinschmidt, 2020-03-05:
Can you ensure your `$RISCV` variable is set to the `chipyard/riscv-tools-install` directory and then post a log of the verilator build that fails?

rayjyh, 2020-03-05:
> ~/Desktop/chipyard/sims/verilator$ echo $RISCV
> /home/jyh/Desktop/chipyard/riscv-tools-install
> ~/Desktop/chipyard/sims/verilator$ 
> ~/Desktop/chipyard/sims/verilator$ 
> ~/Desktop/chipyard/sims/verilator$ make
> make VM_PARALLEL_BUILDS=1 -C /home/jyh/Desktop/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig -f VTestHarness.mk
> make[1]: Entering directory '/home/jyh/Desktop/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig'
> g++  -I.  -MMD -I/home/jyh/Desktop/chipyard/sims/verilator/verilator_install/install/share/verilator/include -I/home/jyh/Desktop/chipyard/sims/verilator/verilator_install/install/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O1 -std=c++11 -I/include -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -I/home/jyh/Desktop/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig -include /home/jyh/Desktop/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.plusArgs -include /home/jyh/Desktop/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/verilator.h -include /home/jyh/Desktop/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig/VTestHarness.h   -c -o SimDTM.o /home/jyh/Desktop/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/SimDTM.cc
> /home/jyh/Desktop/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/SimDTM.cc:3:10: fatal error: fesvr/dtm.h: No such file or directory
>  #include <fesvr/dtm.h>
>           ^~~~~~~~~~~~~
> compilation terminated.
> VTestHarness.mk:69: recipe for target 'SimDTM.o' failed
> make[1]: *** [SimDTM.o] Error 1
> make[1]: Leaving directory '/home/jyh/Desktop/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig'
> Makefile:101: recipe for target '/home/jyh/Desktop/chipyard/sims/verilator/simulator-example-RocketConfig' failed
> make: *** [/home/jyh/Desktop/chipyard/sims/verilator/simulator-example-RocketConfig] Error 2
> ~/Desktop/chipyard/sims/verilator$ 
>

rayjyh, 2020-03-05:
I tried manually changing all the unfound header files to absolute path. The make failed with:
`/usr/bin/ld: cannot find -lfesvr`

colinschmidt, 2020-03-06:
Verilator caches its build directory so its pulling in the old empty `$RISCV` here `-I/include`.
It should fix itself if you do a `make clean` before `make`

KlepD-SAL, 2020-03-09:
Hi,

I also have a problem with building the default example.
The following error occurs:

```
java -Xmx8G -Xss8M -XX:MaxPermSize=256M -jar ~/Documents/chipyard/generators/rocket-chip/sbt-launch.jar ++2.12.10 "project tapeout" "runMain barstools.tapeout.transforms.GenerateTopAndHarness -o ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.top.v -tho ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.harness.v -i ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.fir --syn-top Top --harness-top TestHarness -faf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.anno.json -tsaof ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.top.anno.json -tdf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/firrtl_black_box_resource_files.top.f -tsf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.top.fir -thaof ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.harness.anno.json -hdf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/firrtl_black_box_resource_files.harness.f -thf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.harness.fir --infer-rw --repl-seq-mem -c:TestHarness:-o:~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.top.mems.conf -thconf ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig/example.TestHarness.RocketConfig.harness.mems.conf -td ~/Documents/chipyard/sims/verilator/generated-src/example.TestHarness.RocketConfig"
~/Documents/chipyard/common.mk:68: recipe for target 'firrtl_temp' failed
make: *** [firrtl_temp] Error 137
```

I tried it the most up-to-date version from the repo as well as with version 1.1.0.
It makes no difference. The error is the same.

Any ideas where this is coming from?

OS: Ubuntu 18.04

Cheers

rayjyh, 2020-03-10:
@colinschmidt `make clean` makes no difference.

colinschmidt, 2020-03-11:
These seem to be two separate errors.
One is a failure during verilator build, and the other is a failure to even generate verilog.

@rayjyh Was the error you see the same as before? A failure to find fesvr?

@KlepD-SAL I think you are having the same issue as #455 and we should keep the discussions separate.

KlepD-SAL, 2020-03-11:
@colinschmidt 

I managed to resolve my problems.
My guess is, that there was a problem with the dependencies.
Closely following the [Initial Repository Setup](https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html) made it work, although I think that there is a typo in the Ubuntu dependencies.
**libxpat1-dev** was not available in the repo, so I installed **libexpat1-dev**.
Maybe `scripts/ubuntu-req.sh` should be updated if this really is a typo.

After that I got the same error as @rayjyh:
`No such file or directory #include <fesvr/dtm.h>`

But this happend just because I forgot to launch `env.sh` first (and therefore no $RISCV was set)

So all problems are solved for me.

colinschmidt, 2020-03-11:
@KlepD-SAL Glad to hear that.

Hopefully @rayjyh will be able to resolve their issue similarly.

abejgonzalez, 2020-03-13:
Note, the `libxpat1-dev` error was fixed in #466

alonamid, 2020-04-29:
Closing due to inactivity. Feel free to re-open if there is additional follow-up

closed 2020-04-29 by alonamid