ucb-bar/chipyard #462: SimDRAM fails for 32-bit designs on verilator

jerryz123, 2020-03-06:
**Impact**: rtl

The error is 
```
%Warning-WIDTH: /nscratch/jerryz/chipyard-proj/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RV32RocketConfig/SimDRAM.v:205: Operator ASSIGNDLY expects 32 bits on the Assign RHS, but Assign RHS's VARREF '__r_data' generates 64 bits.
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

VCS has no issue.

I suggest we just add `require(p(XLen) == 64)` when using the BlackBoxSimMem.
Also this may be fixed with Abe's verilator changes?

abejgonzalez, 2020-03-10:
No this was not fixed with my changes. If this needs to be fixed, we can just add a `Wno-fatal` to the compile arguments.

abejgonzalez, 2020-04-29:
I believe this should be fixed with #505